chirpstack/chirpstack-gateway-bridge #2: Question: Can we really infer uplink vs downlink based on PushData vs PullData

ivajloip, 2016-03-24:
Hi,

Really nice job with the packet parser and this bridge (and also the server :smile:)! 

I have a question related to [this][1] one (related to uplink field in PHYPayload). In the current implementation it seems to me that whenever a packet is received on the UDP port and it is PushData, it is assumed to be uplink ([this][2] part of the implementation approximately). I believe it is possible, for example, to receive an *Unconfirmed Data Down* send from another antenna (if two antennas are close enough). In this case the antenna that receives the packet will provide it to the bridge as PushData. In that case I think that the current implementation will not work correctly. Am I misunderstanding the implementation?

If I am correct, I believe that the option with MType is the way to go. The reason for this is that as you said, it should work for every message type except Proprietary and I don't think that there is any guarantee that there will be FHDR in a Proprietary message (or that you can assume anything more than the fact that it will have MHDR and MIC). I might be wrong of course.

Consequently, I would expect Proprietary messages to have only a payload field that is the bytes of the MACPayload or something similar and a custom server to know how to handle this.

Do you have other reasons to prefer the current implementation and does this seems reasonable to you?

[1]: https://github.com/brocaar/lorawan/issues/3
[2]: https://github.com/brocaar/lora-semtech-bridge/blob/master/gateway/backend.go#L364

brocaar, 2016-03-25:
As far as I know there is no way you'll receive a packet transmitted by an other gateway. This is because the "polarization inversion": https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT#L346. This is set to ``true`` on downlink, and is ``false`` on uplink.

ivajloip, 2016-03-25:
According to one of my associates who is working on an experimental LORA network in Rennes, France for more than a year, it is absolutely possible to receive packet from other antennas and this happens to him quite often. Unfortunately, I cannot check this myself. As for "polarization inversion", I did not find references that explain it and so I don't know what it is. 

I guess we have two options: either we just leave things as they are, keeping in mind that we might need to change them some day (if this problem become apparent), or change the code to use the MType and maybe add special behaviour for *Proprietary* (using Push data/Pull data info).

brocaar, 2016-03-25:
As far as I understand the "polarization inversion" (which is documented as "Invert IQ for communications" in the RN2483 command reference) it inverts the transmitted bits, to avoid that gateways receive data from other gateways (as well, they will receive their own data when you don't set the ``ipol`` to ``true``).

I'm not sure if this only works for ``LORA`` modulation or for ``FSK`` as well, I need to do some testing to find out. Until now, I only tested with the ``LORA`` modulation.

If you have a RN2483, you can toggle the polarization with ``ratio set iqi on|off``.

For now I prefer to wait with making any changes related to this, until we are able to reproduce the case where gateways are receiving data down packets.

ivajloip, 2016-03-25:
Thanks for the quick replies. That's fair enough. I will try to learn more about the particular setup and to understand if there is any misconfiguration or any other possible cause.

ivajloip, 2016-03-26:
I spoke with another guy that have a considerable experience with Lora networks and he said that indeed, if you set the `ipol` field, you should not receive the packets on other antennas. If you don't, you might receive your packets and this is most probably what is happening my associate. Thank you very much for explaining this to me!

closed 2016-03-26 by ivajloip

brocaar, 2016-04-12:
FYI: https://github.com/brocaar/lorawan/compare/remove_uplink?expand=1

darkfader, 2020-04-01:
I was quite surprised when I saw an UPLINK / JoinAccept combination. (polarizationInversion was false btw). Someone is trying to be funny probably...