chirpstack/chirpstack-gateway-bridge #6: Multitech conduit issue

arnaudM2X, 2016-06-13:
Thanks for your work. 
I try to use your projet with Multitech Conduit in packetforwarder, but it doesn't work. 
I have messages like this : 
time="2016-06-13T09:48:43+02:00" level=error msg="gateway: could not handle pack
et: gateway: invalid protocol version" addr=192.168.1.93:58730 data_base64=AflXA
gCAAAAAALbF
I need help !!!
Arnaud

brocaar, 2016-06-29:
Sorry for my late reply, I totally missed this issue. In case you haven't solved it yet this is probably the solution: https://github.com/brocaar/lora-semtech-bridge/blob/master/README.md#compatibility

Please check which packet_forwarder you're using and make sure you're using the correct LoRa Semtech Bridge.

arnaudM2X, 2016-06-30:
Thanks for your reply. 
It's look work better with the 1.1.4. Now I need to work with the server (interaction between server and my lora module). Actually I have tow different lora module. One send data but there is en error message on the server and for the second, the is no message on the bridge console.
Thanks again for you help .

arnaudM2X, 2016-06-30:
Sory but now i have this message with the second module : 
could not handle packet : gateway : could not unmarshal PHYPayload : larawan : invalid CID 10

and this other message : 
 could not handle packet : gateway : invalid CRC -1

arnaudM2X, 2016-06-30:
With the 2.01 version i have allready the error message on protocol version

brocaar, 2016-06-30:
I guess the error message explains it all :-) The first packet is invalid since there is no CID 10 defined by the LoRaWAN specs (and it is not in the range of proprietary CIDs). Please see the LoRaWAN specs:

[image]

The second one is rejected because the CRC is invalid. This validation is actually done on the gateway / packet_forwarder and the CRC status is forwarded by the packet_forwarder.

arnaudM2X, 2016-07-01:
Thanks you for your reply. I will continue my investigation with this elements.

[image]<https://cloud.githubusercontent.com/assets/165497/16497591/17de1b20-3ef8-11e6-98b8-2136f8c57f99.png>

closed 2016-09-03 by brocaar