chirpstack/chirpstack-gateway-bridge #9: Support for The Things Network backend

htdvisser, 2016-07-07:
The upcoming version of The Things Network's backend will use gRPC streams to communicate with gateways. The protocol is defined in [this .proto](https://github.com/TheThingsNetwork/ttn/blob/refactor/api/router/router.proto).

I have started implementing an alternative to the `mqttpubsub` backend that will use these gRPC streams in order to bridge to The Things Network.

brocaar, 2016-07-22:
Hi @htdvisser could you tell me what the reason is not to use MQTT? In my opinion gRPC makes the architecture a bit more complex. In the current case, MQTT will take care that the downlink packet is routed to the responsible gateway-bridge, since each gw-bridge is subscribing to the gateway MAC addresses it is responsible for (and will un-subscribe when the gateway disconnects). This way, you can communicate with the gateway(s) without knowing which gw-bridge instance is responsible for which gateway connection.

htdvisser, 2016-07-25:
Our goal was to have direct, streaming transport between each of the distributed network components without the need for external services such as MQTT brokers. For this reason, we use gRPC as the internal communication protocol in The Things Network. Eventually we might implement gRPC directly in the packet forwarders that run on the gateways, but until then, we are using the lora-gateway-bridge to bridge between the two protocols.

brocaar, 2016-07-28:
Ah okay, I understand. Since this backend will be really specific to TTN, I think it might be best to implement this backend (for now) in your own fork (https://github.com/TheThingsNetwork/lora-gateway-bridge). What do you think?

htdvisser, 2016-07-29:
Already did that ;)

closed 2016-07-29 by htdvisser