chirpstack/chirpstack-gateway-bridge #12: Backward compatibility of protocol v1

johanstokking, 2016-07-15:
Many gateways deployed in the field will not be updated any time soon to protocol v2. This means that there's possibly a long overlap of v1 and v2 gateways. This would mean a) that we'd need to have two endpoints for both protocols, which is a configuration setting on the gateway, while the protocol is built in the software. So, b) when using configuration files from wiki, forums, etc., often with the intention to get the frequency plan right, one should also take care of the endpoint.

Is it feasible to make version 2.x of this bridge backward compatible with the v1 protocol? The first byte of the messages indicate the version, so it seems to be a matter of deserializing the payload.

brocaar, 2016-07-28:
Yes, I might consider this in a next version of the bridge. Please note that the new packet_forwarder (using protocol v2) has an important fix; the JIT scheduler. The old packet_forwarder simply overwrites an already scheduled packet when it hasn't been transmitted yet by the radio.

johanstokking, 2016-07-29:
OK thanks. @htdvisser made a [work-around](https://github.com/TheThingsNetwork/lora-gateway-bridge/commit/bb572820b29857dd213a057362b3e8798281dae9) already.

The JIT is indeed is definitely super useful, but it doesn't mean that we can update all gateways. There are many gateways in the field that are a) deployed and not easily updatable and b) use the USB interface and not the SPI, so JIT is not possible (including MultiTech Conduit).

closed 2016-07-29 by johanstokking