chirpstack/chirpstack-gateway-bridge #13: Error gateway: could not handle packet: parsing time ""2011-01-01T06:04:25Z"" as ""2006-01-02 15:04:05 MST"

shmrymbd, 2016-08-19:
INFO[0000] starting LoRa Gateway Bridge                  docs=https://docs.loraserver.io/lora-gateway-bridge/ version=2.1.0
INFO[0000] backend: connecting to mqtt broker            server=tcp://127.0.0.1:1883
INFO[0000] gateway: starting gateway udp listener        addr=0.0.0.0:1700
INFO[0000] backend: connected to mqtt broker            
INFO[0004] gateway: received udp packet from gateway     addr=192.168.1.155:43002 protocol_version=1 type=PullData
INFO[0004] backend: subscribing to topic                 topic=gateway/1234567887654321/tx
INFO[0004] gateway: sending udp packet to gateway        addr=192.168.1.155:43002 protocol_version=1 type=PullACK
INFO[0004] gateway: received udp packet from gateway     addr=192.168.1.155:43001 protocol_version=1 type=PushData
ERRO[0004] gateway: could not handle packet: parsing time ""2011-01-01T06:04:25Z"" as ""2006-01-02 15:04:05 MST"": cannot parse "T06:04:25Z"" as " "  addr=192.168.1.155:43001 data_base64=AZh8ABI0VniHZUMheyJzdGF0Ijp7InRpbWUiOiIyMDExLTAxLTAxVDA2OjA0OjI1WiIsInJ4bmIiOjE2LCJyeG9rIjoxMiwicnhmdyI6MTIsImFja3IiOjMyLCJkd25iIjowLCJ0eG5iIjowfX0=

brocaar, 2016-08-20:
The `base64_data` (`AZh8ABI0VniHZUMheyJzdGF0Ijp7InRpbWUiOiIyMDExLTAxLTAxVDA2OjA0OjI1WiIsInJ4bmIiOjE2LCJyeG9rIjoxMiwicnhmdyI6MTIsImFja3IiOjMyLCJkd25iIjowLCJ0eG5iIjowfX0=`) decodes to:

```json
{
	"stat": {
		"time": "2011-01-01T06:04:25Z",
		"rxnb": 16,
		"rxok": 12,
		"rxfw": 12,
		"ackr": 32,
		"dwnb": 0,
		"txnb": 0
	}
}
```

However, the `time` key has a different formatting than is documented in https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT#L231. As well, it seems a little bit in the past 😉 

Could you let me know which version of the packet_forwarder you're using?

shmrymbd, 2016-08-21:
1. Bridge version=2.1.0 i download from your guide
2. my lora gateway using 
 - microchip gateway with sx1301..
 - https://github.com/JaapBraam/LoRaWanGateway esp with inair9b

brocaar, 2016-08-21:
To me it seems that the time format sent by the
https://github.com/JaapBraam/LoRaWanGateway project has an incompatibility
with the packet_forwarder protocol.

jamiechard, 2016-08-25:
Forgive me if I'm stating the obvious, but I "worked around" the above for another gateway by changing the "Time" inside the "type Stat struct" in gateway/structs.go (i.e. line 396) from "ExpandedTime" to "CompactTime".

brocaar, 2016-08-26:
Yes, but I think if you would use the "official" packet_forwarder now (https://github.com/Lora-net/packet_forwarder), it wouldn't be able to parse that timestamp. I think it is better to open a ticket here https://github.com/JaapBraam/LoRaWanGateway and ask them to change their timestamp format according to the [PROTOCOL.TXT](https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT#L231).

closed 2016-09-28 by brocaar

htdvisser, 2016-09-29:
For reference: 
https://github.com/JaapBraam/LoRaWanGateway/commit/01bf6512a6f2d0f3e598de4dd03071b73ae328ff fixed this issue

alexanderplanar, 2017-06-08:
Hello, brocaar!
I've got the same problem with Microchip LoRa Gateway!
Does it mean, that the problem is in gateway firmware and I need update it, or, maybe, there is a several legal time formats of "stat":"time" variable? I tried to form this packet by hand using Wireshark and UDP-Tools, I just replaced '..T...Z' string to '.... GMT' format, and it correctly passes parsing.... 

Now I think about rebuilding the server for my format, because I'm not able to reprogramm the LoRa gateway, cuz I do not have any firmware binaries or sources for this PCB...

sslupsky, 2017-08-15:
@alexanderplanar Did you fix this problem?  I recently installed the gateway_bridge with the Microchip gateway and I am seeing this error.  I am not sure if I can just ignore it?

alexanderplanar, 2017-08-16:
sslupsky, bad news, not yet, I had to put aside my work with LoRa for some time due to the other work. 
But I guess this could not be fixed until either the firmware of Microchip gateway will be upgraded or the gateway bridge source code will be fixed. The last one is not recommended because, as far I understand, it corresponds to the standard, and fixing it could cause other problems with other gateways. So, I'm still ignoring this error, because I do not have the fixed firmware for Microchip gateway.
The last way I see it is to retranslate messages from Microchip gateway to the gateway_bridge software using a UDP-proxy application. I do not know is these application already written by someone, but I'm sure it is not so complicated to create your own UDP-proxy using e.g. GCC and some kind of notepad-editor. The only few bytes needs to be fixed in the message before retranslating it to the software.

brocaar, 2017-08-16:
@alexanderplanar @sslupsky please see https://github.com/brocaar/lora-gateway-bridge/issues/28.

sslupsky, 2017-08-16:
@alexanderplanar @brocaar It is unfortunate that Microchip has not supplied a fix to this problem.  It has been around for a while and likely a trivial fix that would be fixed by now if Microchip open sourced their gateway firmware.  Has anyone asked them to do so?  @brocaar , a request from you would likely get their attention.