chirpstack/chirpstack-gateway-bridge #15: mosquitto_sub -t "gateway/+/rx" time message was not set

yuseunghyuk, 2016-09-02:
lora_pkt_fwd : Version 3.0.0
log_lora-gateway-bridge : Version 2.1.0-3-gedf27bf

when I subscribe gateway's rx
time was not set like this.

"time":"0001-01-01T00:00:00Z"

log file is  below.

[log_lora-gateway-bridge-2.1.0-3-gedf27bf.txt](https://github.com/brocaar/lora-gateway-bridge/files/450955/log_lora-gateway-bridge-2.1.0-3-gedf27bf.txt)
[log_packet_forwarder-3.0.0.txt](https://github.com/brocaar/lora-gateway-bridge/files/450956/log_packet_forwarder-3.0.0.txt)
[log_mqtt_gateway_rx.txt](https://github.com/brocaar/lora-gateway-bridge/files/450957/log_mqtt_gateway_rx.txt)

tftelkamp, 2016-09-02:
Does your gateway have a GPS? If not, lora_pkt_fwd will not send a time field. 

It would be better though if the lora-gateway-bridge would not include the time field either, instead of filling it with this default value.

brocaar, 2016-09-03:
Thanks! I'll look into it after https://github.com/brocaar/loraserver/issues/68.

commit 57cbb70d (2016-09-25): Omit empty time field.

Closes #15.
closed 2016-09-25 by brocaar