chirpstack/chirpstack-gateway-bridge #16: FSK downlink needs fdev set

tftelkamp, 2016-09-02:
Please add fdev to the FSK txpk, with a value of 25000.

ipol you can remove (it only applies to LORA).

JSON down: {"txpk":{"imme":false,"tmst":1082005352,"freq":868.8,"rfch":0,"powe":14,"modu":"FSK","datr":50000,"ipol":false,"size":12,"data":"YFR4RQIgAACSQU77"}}
WARNING: [down] no mandatory "txpk.fdev" object in JSON, TX aborted

tftelkamp, 2016-09-05:
Addendum.

For a (G)FSK modulation index of 1 (as specific in AN1200.24), the frequency deviation (fdev) is half of the bitrate. Hence the 25000.

However, LoRaWAN only specifies an FSK bitrate of 50kbps. So maybe it's better regardless of the uplink datarate to always reply with datarate 50000 and fdev 25000.

htdvisser, 2016-09-23:
See also https://github.com/TheThingsNetwork/lora-gateway-bridge/commit/78d163b68ef818357b837f7f6a15f4d6dd60d624#diff-c1f2b900561bef280eae8703e2a1329cR459

commit 26ebc69f (2016-09-25): Populate FDev field when using FSK.

Closes #16.
closed 2016-09-25 by brocaar