chirpstack/chirpstack-gateway-bridge #18: I can use it ,but i can't make it ,why?

lancezhangsf, 2016-11-02:
Hi,
i want to compile the project ,but some errors happen .
$ make 

Compiling source for linux 
cmd/lora-gateway-bridge/main.go:10:2: import "github.com/Sirupsen/logrus": cannot find package
cmd/lora-gateway-bridge/main.go:11:2: import "github.com/brocaar/lora-gateway-bridge/backend/mqttpubsub": cannot find package
cmd/lora-gateway-bridge/main.go:12:2: import "github.com/brocaar/lora-gateway-bridge/gateway": cannot find package
cmd/lora-gateway-bridge/main.go:13:2: import "github.com/brocaar/lorawan": cannot find package
cmd/lora-gateway-bridge/main.go:14:2: import "github.com/codegangsta/cli": cannot find package
make: *** [build] Error 1

a-farooq, 2016-11-08:
/////create folder as following
# mkdir -p gw-bridge/src/github.com/brocaar/
# cd gw-bridge
# export GOPATH=$(pwd)
# cd src/github.com/brocaar/
/////clone the project here
/////name this folder as lora-gateway-bridge and then compile
# make build

brocaar, 2016-11-13:
@lancezhangsf make sure you're using the latest Go version or that the `GO15VENDOREXPERIMENT` env variable is set.

closed 2016-12-27 by brocaar