chirpstack/chirpstack-gateway-bridge #19: Time is always "0001-01-01T00:00:00Z" in MQTT messages (using a LORANK8)

glederer007, 2016-11-03:
Since a couple of weeks, I noticed that the "time" parameter at the rx packet (in both "gateway" and "application" MQTT packets) is always "time":"0001-01-01T00:00:00Z". I must say that this can be seen since I upgraded my Lorank8, so probably the problem could be with the Lorank software as well. To double-check I started a UDP listener to see if the UDP packets contain time as "zero" as well. Well, the UDP packets that are sent to the lora-gateway-bridge apparently contain the correct time (i.e. "time":"2016-10-27T15:16:32Z","rxpk" ... ). Is this a bug?

A sample UDP dump is attached here: 
[UDP_20161103-1212.txt](https://github.com/brocaar/lora-gateway-bridge/files/568784/UDP_20161103-1212.txt)

I also include the MQTT payload that is generated by the lora-gateway-bridge from the UDP message with timestamp 8821235:
[MQTT_8821235.txt](https://github.com/brocaar/lora-gateway-bridge/files/568785/MQTT_8821235.txt)

brocaar, 2016-11-13:
This is because the `time` field is only filled in the UDP packet when the gateway has an accurate time source (GPS). Since an empty time in Go is equal to `0001-01-01T00:00:00Z`, the JSON published to MQTT is filled with this value. This has been brought up before and I agree it is confusing. I'll look into a proper solution soon.

glederer007, 2016-11-13:
Thanks for your reply. Don't worry too much about it, it is really not a big issue.

jpmeijers, 2017-01-23:
How about setting the time field to 1970-01-01 00:00 (unix epoch)?

On the other hand, if the time is not set because the gateway is not aware of the current time, how about setting it to the current timestamp of the server. If the server is NTP sync'ed this will at least give a good representation of when the packet was passed through the bridge.

tftelkamp, 2017-01-23:
Please don't set it to current time in the server, because then we don't know anymore of this is GPS synced time from the gateway or not.

Server time is useful though, but then better add a field for it.

brocaar, 2017-01-27:
I don't thing it would be a good idea to let LoRa Gateway Bridge add a time-field to the JSON. As the instance might not run on the gateway itself, there might be quite a difference between when it was received by the gateway and when it was received by the gw bridge (as in seconds in some setups using 3G connection).

mikumaycry, 2017-04-05:
I got the same issue.My gateway doesn't have a GPS, but it could sync time from the Internet.
The packet forwarder on my Raspberry Pi 3 logs correct timestamp. But timestamp is transformed into a wired value after data is transferred by gate-way-bridge.

sslupsky, 2017-08-22:
I have seen this issue when using the Microchip gateway as well so I do not think this is specific to the LORANK8.

Here is the output when subscribed to "gateway/#/#":

gateway/1234567887654321/rx {"rxInfo":{"mac":"1234567887654321","time":"0001-01-01T00:00:00Z","timestamp":1433980587,"frequency":903100000,"channel":4,"rfChain":1,"crcStatus":1,"codeRate":"4/5","rssi":-39,"loRaSNR":9.8,"size":21,"dataRate":{"modulation":"LORA","spreadFactor":7,"bandwidth":125}},"phyPayload":"QHC9cwcAIgI8qBIoYzAPfh8iXsvN"}

brocaar, 2017-12-11:
This will be fixed soon :-)

commit 485fbe87 (2017-12-13): Implement TX ACK and RX time / tmms field.

Closes #19.
closed 2017-12-13 by brocaar