chirpstack/chirpstack-gateway-bridge #20: where to run packet forwarder

a-farooq, 2016-11-08:
Hi Brocaar,

I have done the setup given by you.
I m having Microchip gateway and motes. I m running the lora-gateway-bridge, loraserver, lora-app-server shared by you.

I m getting the packets from microchip gateway to the lora gateway bridge. 
But I m getting the following error.
=============
Error gateway: could not handle packet: parsing time 
=============

You said that running the three software only was required for the setup. Do I have to run packet forwarder too ?
If yes, where?

P.S. this question was already asked but it was closed and it did not solve my query.

brocaar, 2016-11-13:
Could you tell me which `packet_forwarder` version you're using on your gateway?

dhavalp, 2016-12-02:
Hi,
I have setup the lora server now I want to receive the data sent from MicroChip Lora Mote (RN2483) to the lora gateway bridge how to do that? If you can guide me. 

Thanks

closed 2016-12-27 by brocaar