chirpstack/chirpstack-gateway-bridge #28: Parsing time error

gibosler, 2017-02-16:
Hi Broccar, 
we are in the testing phase of  Lora server with Microchip Lora Gatwey version 1.0.4.
We had this error: 

feb 16 23:02:29 ******** lora-gateway-bridge[12324]: time="2017-02-16T23:02:29+01:00" level=error msg="gateway: could not handle packet: parsing time \"\"2017-02-16T22:01:57Z\"\" as \"\"2006-01-02 15:04:05 MST\"\": cannot parse \"T22:01:57Z\"\" as \" \"" addr=****:43001 data_base64=AXEKABI0VniHZUMheyJzdGF0Ijp7InRpbWUiOiIyMDE3LTAyLTE2VDIyOjAxOjU3WiIsInJ4bmIiOjc4LCJyeG9rIjozMiwicnhmdyI6MzIsImFja3IiOjEsImR3bmIiOjIwLCJ0eG5iIjoyMH19

Could you please help us to solve this problem?
Thank you in advance.
Regards

Gilberto

brocaar, 2017-02-17:
Hi @gibosler Thanks for reporting the issue, I'm already aware of it! There are some difference in the UDP format that the packet_forwarder sends and how the Microchip gateway sends the data.

Good news however is that soon I'll have a Microchip gateway board to test with (track&trace says that it will be delivered today!)

brocaar, 2017-02-18:
I have a Microchip gateway board and did some testing. It seems that not only the timestamp is wrong, there are some more issues with the format of the gateway stats (the 'normal' packet_forwarder is sending deltas, Microchip is sending the total since the gateway was started).

Note that this issue only affects the collection of stats, sending and receiving LoRa frames should not be affected by this issue.

acourt, 2017-04-14:
I am experiencing this exact issue as well. It seems like this causes the gateway to not automatically get registered by the loraserver. Will manually adding the gateway through the API solve the problem?

I will try to send uplink/downlink data.

Are you still planning on doing a workaround or should I order a new gateway?

brocaar, 2017-04-15:
I'm not planning to do the workaround as it would not solve the issue. The stats sent by the gateway are also incorrect (which I can't work around). I've notified Microchip about this issue a few months ago, so far they haven't fixed it...

epolicar, 2017-06-03:
I am having this issue also. However, the Microchip gateway (just bought, by the way) works with The Things Network server. Since TTN's server limits the allowed downlinks to only 10 per day, and does not support Class C, I thought I give your server a try. Does this imply that TTN also does not follow the standard? Or could it be they adjust their server config to cater to a particular gateway?

I remember reading a similar issue from some thread in this lora-gateway-bridge forum but I can't locate it now. It's about someone changing a setting from "Expanded" to "Compact". He claims it works. Can you elaborate on this? If it does work, then it can be a temporary measure until Microchip fixes the problem.

I must say you did an impressive job. Keep up the good work!

brocaar, 2017-06-19:
@epolicar, please note that the gateway **does** work with LoRa Server, only the gateway statistics pushed by the gateway are ignored because of the incorrect data format.

freebeans, 2017-06-19:
I'm also testing this gateway with LoRa Server, however, even though the lora-gateway-bridge keeps sending the packets through MQTT, the gateway never shows up on the lora-app-server interface. I tried registering it's MAC, but it shows no activity.

Is this anyhow related to the fact that the lora-gateway-bridge claims it cannot parse the timestamp?

brocaar, 2017-06-19:
Yes that is correct. As it is unable the parse the stats received by the gateway, it can't show any activity.

freebeans, 2017-06-19:
Can you please reference me to the document that details the time format that should be used?
I need to set up a demo with this gateway and I don't mind modifying your code to accept this just for that demo.

brocaar, 2017-06-19:
Please note that the gateway does work, only you won't see any statistics. For the format of these stats packets please refer to https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT#L232. Also note that even if you fix the code at the LoRa Server side, the stats are still wrong as the Microchip firmware does not aggregate per interval, instead it just continues counting. 

For your demo you could generate some stats yourself too, by publishing fake stats to the `gateway/[mac]/stats` MQTT topic (https://docs.loraserver.io/lora-gateway-bridge/use/data/).

freebeans, 2017-06-19:
Thank you. It seems my gateway stops receiving packets after some time. My guess is it's not receiving acknowledge for the stats and is somewhat freezing. I'll try to implement something. Thanks.

brocaar, 2017-11-19:
I'm going to close this issue. I've raised this bug long time ago to Microchip, but it doesn't seem they are going to fix this.

closed 2017-11-19 by brocaar