chirpstack/chirpstack-gateway-bridge #29: lora-gateway-bridge can't receive the packet from mote.

SoHyunYang, 2017-02-23:
I run packet_forwarder and lora-gateway-bridge at the same time and let the LoRa mote send packet to the gateway.
(I used the iC880A as the gateway and the iM880B Starter kit as the mote.)
However, packet_fowrder confirmed that the packet was received, but the lora-gateway-bridge did not respond.

I thought of a problem. But I do not know if this caused this error. If it seems correct, please let me know about the values' ​​setting related server in global_conf.json file of packet forwarder.
[image]
[image]

I was setting the global_conf.json file of pacekt_forwarder / lora_pkt_fwd. (Server_address: mqtt broker's tcp address, serv_port_up, serv_port_down: mqtt broker's port). But I think it is not right.

I wonder if there might be other problems.

Thank you.

brocaar, 2017-02-24:
By default LoRa Gateway Bridge uses port 1700 for up and down. That might fix your issue.

SoHyunYang, 2017-02-25:
ok. The defalut udp-bind value is port 1700. And I confirmed the port 1700's state is not "listen" or "established" when i am running lora-gateway-bridge. How can i do to solve this problem? 
[image]

The following information is the Installation Environment set by me.

lora app server, lora server, mqtt broker : in window
mqtt  tcp : 192.168. 0.11 , port :1883 
window ip : 192.168.0.11

lora-gateway, packet forwarder, lora-gateway-bridge : in raspberrypi connected with iC880A concentrator(gateway)
raspberrypi ip : 192.168.0.12

Thank you.

closed 2017-03-05 by SoHyunYang