chirpstack/chirpstack-gateway-bridge #30: No /etc/lsb-release under Debian

Champal, 2017-03-10:
Hi, 

Here https://docs.loraserver.io/lora-gateway-bridge/getting-started/#setting-up-lora-gateway-bridge you 
use the file `/etc/lsb-release` that does not exist under Debian Jessie,  you have to use the `lsb_release` command instead.

```
source /etc/lsb-release
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1CE2AFD36DBCCA00
sudo echo "deb https://repos.loraserver.io/${DISTRIB_ID,,} ${DISTRIB_CODENAME} testing" | sudo tee /etc/apt/sources.list.d/loraserver.list
sudo apt-get update
```
You can replace `source /etc/lsb-release` by 
```
export DISTRIB_ID=`lsb_release -si`;export DISTRIB_CODENAME=`lsb_release -sc`
```

brocaar, 2017-03-15:
Thanks @Champal, I'll fix this in the documentation.

andykv59, 2017-05-08:
Will appreciate for the fix

closed 2017-05-16 by brocaar

linas, 2019-03-22:
FWIW totally off-topic, but on my machine things were broken because `/etc/dpkg/origins/default -> /etc/dpkg/origins/ubuntu` instead of `/etc/dpkg/origins/default -> /etc/dpkg/origins/debian` like it should have been... and that broke lsb_release with cryptic python errors.

hpsaturn, 2020-03-24:
> Hi,
>
> Here https://docs.loraserver.io/lora-gateway-bridge/getting-started/#setting-up-lora-gateway-bridge you
> use the file `/etc/lsb-release` that does not exist under Debian Jessie, you have to use [...]

Thanks. You can use in Debian:

```bash
source /etc/os-release
```

commit 10e32f0a (2017-05-16): Fix no /etc/lsb-release under Debian issue.

Closes #30.