chirpstack/chirpstack-gateway-bridge #35: Kerlink iFemtoCell does not send (or lora-gateway-bridge does not receive) timestamp

Vini74, 2017-06-06:
Kerlink iFemtoCell with Semtech Packet Forwarder does not send (or lora-gateway-bridge does not receive) timestamp.

Packet from Kerlink iFemtoCell:

{"applicationID":"1","applicationName":"0000000000000077","nodeName":"0018b20000000298","devEUI":"0018b20000000298",
"rxInfo":[{"mac":"7276ff00390301cb","rssi":-48,"loRaSNR":6.5,"name":"Kerlink-Home-2","latitude":55.776826451225254,"longitude":37.730076313018806,"altitude":0}],
"txInfo":{"frequency":868500000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":12},"adr":true,"codeRate":"4/5"},"fCnt":347,"fPort":1,"data":"nx1VRmBgA3Q3kFxbDOA6Bw=="}

Packet from Kerlink LoRa station (Wirnet Station):

{"applicationID":"1","applicationName":"0000000000000077","nodeName":"0018b20000000298","devEUI":"0018b20000000298","rxInfo":[{"mac":"0000024b080e0e0c","time":"2017-06-06T13:45:08.343905Z","rssi":-15,"loRaSNR":9.5,"name":"KERLINK-GTL-EU","latitude":55.77681,"longitude":37.73001,"altitude":89}],"txInfo":{"frequency":868300000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":12},"adr":true,"codeRate":"4/5"},"fCnt":388,"fPort":1,"data":"nh1VRlmQA3Q3kIV9DOA="}

Vini74, 2017-06-06:
This is UDP dumps from both gateways

KERLINK iFemtoCell

\02|\c2\00rv\ff\009\03\01\cb{"rxpk":[{"tmst":38690196,"chan":6,"rfch":1,"freq":868.300000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":9.8,"rssi":-55,"size":27,"data":"gAHe3t6A6QEBreNWenV8IrHbytfFOxNXaZ3R"}]}

\02\9b\b4\00rv\ff\009\03\01\cb{"rxpk":[{"tmst":118689572,"chan":5,"rfch":1,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":9.8,"rssi":-46,"size":27,"data":"gAHe3t6A6gEBNnjozjVRsC+o0P0TRhO1MnRI"}]}

KERLINK OUTDOOR

\01J\ec\00\00\00\02K\08\0e\0e\0c{"rxpk":[{"tmst":25766268,"chan":1,"rfch":1,"freq":868.300000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":9.0,"rssi":-10,"size":27,"data":"gAHe3t6A8QEBo25xEPXG7S7tamAYotw3lDzr"}]}

\013\9f\00\00\00\02K\08\0e\0e\0c{"rxpk":[{"tmst":65765940,"chan":2,"rfch":1,"freq":868.500000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":8.2,"rssi":-10,"size":29,"data":"gAHe3t6A8gEB5nzMqpx1cPMIBbhqcYCfL1b5mLo="}]}

\01!=\00\00\00\02K\08\0e\0e\0c{"rxpk":[{"tmst":125765492,"chan":0,"rfch":1,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":10.0,"rssi":-33,"size":29,"data":"gAHe3t6A8wEBQdNOB6SHrqolgbWfN59j4sfILDc="}]}

zhladik, 2017-06-13:
Hello, 
I just got iFemtoCell to my desk, no doc, no binaries, did you succesfully compilesrc form Github repo?
May be missing time is because fw expect GPS type info?

brocaar, 2017-06-19:
When the iFemtoCell does not have a GPS, then the above is expected behaviour.

Vini74, 2017-06-19:
[image]
iFemtoCell does not have a GPS at all.
In that case, where do I get the payload timestamp?

brocaar, 2017-06-19:
You don't get any timestamp in case the gateway does not have a GPS.

zhladik, 2017-06-19:
but with Loriot Blob GW app time is there. In fact GPS acurate time may be usable for localisation purposes, In other case NTP based time (and iFemtocell does run ntpd) is acurate enough. So may be code for ntp time use is by mistake disabled for iFemtocell. Timestamp is usualy quite usable info (and sensors usualy do not have RTC and does not send time in payload).

Vini74, 2017-06-19:
Next question. why I receive latitude and longitude?
application/1/node/0018b20000000298/rx
{"applicationID":"1","applicationName":"0000000000000077","nodeName":"0018b20000000298","devEUI":"0018b20000000298","rxInfo":[{"mac":"7276ff00390301cb","rssi":-39,"loRaSNR":7.5,"name":"Kerlink-Home-2","latitude":55.776826451225254,"longitude":37.730076313018806,"altitude":0}],"txInfo":{"frequency":868500000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":12},"adr":true,"codeRate":"4/5"},"fCnt":80,"fPort":1,"data":"jxxRUAzhMAk="}

brocaar, 2017-06-21:
I guess because you configured the lat / longitude in LoRa App Server :-)

closed 2017-07-15 by brocaar