chirpstack/chirpstack-gateway-bridge #38: OTAA Node - does not get activated

nibelungen, 2017-06-14:
Hi,

thanks to @brocaar for this great piece of work!

I tried to connect a lopy node with OTAA (https://github.com/pycom/pycom-libraries/blob/master/examples/lorawan-nano-gateway/otaa_node.py) to a single channel gateway (https://github.com/tftelkamp/single_chan_pkt_fwd).
So far connection between node - gateway and gateway-bridge works fine, but i think the node does not get the acitvation message it still sends join requests and the gateway bridge always replies with join accept:

[image]

That happens multiple time each minute:

[image]

Did is miss something? Maybe some one use the same libraray and has success?

Loraserver Loraappserver and Loragatewaybridge AND single_chan_pkt_fwd run on the same rpi.

Thanks in advance,
best regards
Andreas

siscia, 2017-06-16:
Definitely I have no idea of what is going on, however.

Do you run also the MQTT server somewhere?

nibelungen, 2017-06-19:
All is running on the same rpi. I think the single chan pkt fwd is not capable of OTAA due to the lack of TX function? 

From the Readme:
Features:
listen on configurable frequency and spreading factor
SF7 to SF12
status updates
can forward to two servers

Not (yet) supported:
PACKET_PUSH_ACK processing
SF7BW250 modulation
FSK modulation
downstream messages (tx)

brocaar, 2017-06-21:
Not (yet) supported: **downstream messages (tx)**. That looks like your issue 😉

nibelungen, 2017-06-21:
Thanks for your reply @brocaar :)
Thats my last suggestion too - i switched ABP as a workaround but i now stuck at the received data... It looks like it is encrypted, im using the pycom lora node code.
So i have to search for a solution. But for this issue it is definitly no lora-gateway-bridge issue!

brocaar, 2017-06-22:
Please note that all data is published decrypted to MQTT by LoRa App Server https://docs.loraserver.io/lora-app-server/integrate/data/. However you need to base64 decode this.

closed 2017-06-22 by brocaar