chirpstack/chirpstack-gateway-bridge #41: question: listening ipv6 instead of ipv4?

PhatHub, 2017-06-19:
When I try to start the Lora gateway bridge, the UDP listener ends up creating an ipv6 service instead of ipv4. I've tried to force it by editing `backend.go` (approx line 100) to have:

`conn, err := net.ListenUDP("udp4", addr)`

However, `netstat -lun` still shows it as listening only via udp6:

```
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State      
udp        0      0 127.0.0.1:323           0.0.0.0:*                          
udp        0      0 0.0.0.0:32683           0.0.0.0:*                          
udp        0      0 0.0.0.0:68              0.0.0.0:*                          
udp6       0      0 ::1:323                 :::*                               
udp6       0      0 :::38377                :::*                               
udp6       0      0 :::1700                 :::*   
```

Is that normal?

brocaar, 2017-06-21:
I have the same here, but all works fine with ipv4. This method is used to "resolve" the UDP address: https://golang.org/pkg/net/#ResolveUDPAddr.

PhatHub, 2017-06-21:
Ok! Thanks, I just wanted to make sure!

closed 2017-06-21 by PhatHub