chirpstack/chirpstack-gateway-bridge #44: Kerlink Wirnet Dual Antenna Support

finnoliver, 2017-07-04:
Hi,
the Kerlink Wirnet iBTS seems to struct the rx Data a little different (see UDP Dump below), when using two antennas and standart pkt-forwarder. That way the gateway bridge doesn't seem to be able to handle RSSI and SNR correctly, when carrying over to MQTT. Would it be possible to  add support for that ? 

Greetings
Finn

`{"rxpk":[{"tmst":879148780,"time":"2017-07-04T13:51:17.997099Z","rfch":0,"freq":868.500000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","size":24,"data":"gM+AMQcAvgQBlohnlJqUGOJKTDuTscQD","rsig":[{"ant":0,"chan":7,"rssic":-95,"lsnr":14.0,"etime":"42QMzOlYSSPMMeqVPrY0fQ==","rssis":-95,"rssisd":0,"ftime":1255738435,"foff":-8898,"ft2d":-251,"rfbsb":100,"rs2s1":97},{"ant":1,"chan":23,"rssic":-88,"lsnr":14.0,"etime":"djGiSzOC+gCT7vRPv7+Asw==","rssis":-88,"rssisd":0,"ftime":-1252538435,"foff":-8898,"ft2d":-187,"rfbsb":100,"rs2s1":104}]}]}`

brocaar, 2017-07-06:
Thanks! I'll have to dig into this as it is not mentioned in the packet-forwarder [PROTOCOL.TXT](https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT). Which version of the packet-forwarder does this gateway run?

finnoliver, 2017-07-06:
Hi brocaar, thanks for answering. The Protocol.txt included in the package is allegedly 1.0, which I guess is pretty old and a bit weired. I'll attach the file. The package forwarder itself is supposed to be v3.1 and there are changes to the.
[PROTOCOL.TXT](https://github.com/brocaar/lora-gateway-bridge/files/1128419/PROTOCOL.TXT)

finnoliver, 2017-07-07:
#45 I tried to write a little patch.

commit a96b4670 (2017-11-25): Add iBTS compatibility.

Note: a frame received by multiple antennas is emitted as multiple JSON over MQTT messages.

Closes #44.
commit 80ec8177 (2017-12-01): Add iBTS compatibility.

Note: a frame received by multiple antennas is emitted as multiple JSON over MQTT messages.

Closes #44.
PR #68 (2017-12-01): Add iBTS compatibility.
Note: a frame received by multiple antennas is emitted as multiple JSON over MQTT messages.

Closes #44.
closed 2017-12-01 by brocaar