chirpstack/chirpstack-gateway-bridge #48: sudo-apt get update error.

monkmahadeva, 2017-08-02:
### Is this a bug or a feature request?

### What did you expect?

### What happened?

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?

monkmahadeva, 2017-08-02:
During sudo apt-get update Its ignoring most of the packages. Here's the log 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
redis-server is already the newest version (2:3.0.6-1).
redis-tools is already the newest version (2:3.0.6-1).
postgresql is already the newest version (9.6+184.pgdg16.04+1).
The following package was automatically installed and is no longer required:
  ubuntu-core-launcher
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  libc-ares2 libev4 libmosquitto1 libuv1 libwebsockets7
The following NEW packages will be installed:
  libc-ares2 libev4 libmosquitto1 libuv1 libwebsockets7 mosquitto mosquitto-clients
0 upgraded, 7 newly installed, 0 to remove and 23 not upgraded.
Need to get 179 kB/418 kB of archives.
After this operation, 1,054 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://in.archive.ubuntu.com/ubuntu xenial-updates/main amd64 libc-ares2 amd64 1.10.0-3ubuntu0.1 [34.1 kB]
Get:2 http://in.archive.ubuntu.com/ubuntu xenial/universe amd64 libuv1 amd64 1.8.0-1 [57.4 kB]                                                
Get:3 http://in.archive.ubuntu.com/ubuntu xenial/universe amd64 libev4 amd64 1:4.22-1 [26.3 kB]                                               
Get:4 http://in.archive.ubuntu.com/ubuntu xenial/universe amd64 libwebsockets7 amd64 1.7.1-1 [61.0 kB]                                        
Fetched 179 kB in 44s (3,975 B/s)                                                                                                             
Selecting previously unselected package libc-ares2:amd64.
(Reading database ... 174540 files and directories currently installed.)
Preparing to unpack .../libc-ares2_1.10.0-3ubuntu0.1_amd64.deb ...
Unpacking libc-ares2:amd64 (1.10.0-3ubuntu0.1) ...
Selecting previously unselected package libmosquitto1:amd64.
Preparing to unpack .../libmosquitto1_1.4.14-0mosquitto1~xenial1_amd64.deb ...
Unpacking libmosquitto1:amd64 (1.4.14-0mosquitto1~xenial1) ...
Selecting previously unselected package libuv1:amd64.
Preparing to unpack .../libuv1_1.8.0-1_amd64.deb ...
Unpacking libuv1:amd64 (1.8.0-1) ...
Selecting previously unselected package libev4.
Preparing to unpack .../libev4_1%3a4.22-1_amd64.deb ...
Unpacking libev4 (1:4.22-1) ...
Selecting previously unselected package libwebsockets7:amd64.
Preparing to unpack .../libwebsockets7_1.7.1-1_amd64.deb ...
Unpacking libwebsockets7:amd64 (1.7.1-1) ...
Selecting previously unselected package mosquitto.
Preparing to unpack .../mosquitto_1.4.14-0mosquitto1~xenial1_amd64.deb ...
Unpacking mosquitto (1.4.14-0mosquitto1~xenial1) ...
Selecting previously unselected package mosquitto-clients.
Preparing to unpack .../mosquitto-clients_1.4.14-0mosquitto1~xenial1_amd64.deb ...
Unpacking mosquitto-clients (1.4.14-0mosquitto1~xenial1) ...
Processing triggers for libc-bin (2.23-0ubuntu9) ...
Processing triggers for man-db (2.7.5-1) ...
Processing triggers for ureadahead (0.100.0-19) ...
Processing triggers for systemd (229-4ubuntu19) ...
Setting up libc-ares2:amd64 (1.10.0-3ubuntu0.1) ...
Setting up libmosquitto1:amd64 (1.4.14-0mosquitto1~xenial1) ...
Setting up libuv1:amd64 (1.8.0-1) ...
Setting up libev4 (1:4.22-1) ...
Setting up libwebsockets7:amd64 (1.7.1-1) ...
Setting up mosquitto (1.4.14-0mosquitto1~xenial1) ...
Setting up mosquitto-clients (1.4.14-0mosquitto1~xenial1) ...
Processing triggers for libc-bin (2.23-0ubuntu9) ...
user@user-X550LC:/$ sudo apt-get update
Hit:1 http://security.ubuntu.com/ubuntu xenial-security InRelease                                                                             
Hit:2 http://ppa.launchpad.net/mosquitto-dev/mosquitto-ppa/ubuntu xenial InRelease
Hit:3 http://in.archive.ubuntu.com/ubuntu xenial InRelease              
Hit:4 http://in.archive.ubuntu.com/ubuntu xenial-updates InRelease
Hit:5 http://in.archive.ubuntu.com/ubuntu xenial-backports InRelease
Hit:6 https://repos.loraserver.io/ubuntu xenial InRelease      
Ign:7 https://repos.loraserver.io/ubuntu xenail InRelease       
Ign:8 https://repos.loraserver.io/ubuntu xenail Release
Ign:9 https://repos.loraserver.io/ubuntu xenail/testing amd64 Packages
Ign:10 https://repos.loraserver.io/ubuntu xenail/testing i386 Packages
Ign:11 https://repos.loraserver.io/ubuntu xenail/testing all Packages
Ign:12 https://repos.loraserver.io/ubuntu xenail/testing Translation-en_IN
[... 26 lines cut ...]
Ign:10 https://repos.loraserver.io/ubuntu xenail/testing i386 Packages
Ign:11 https://repos.loraserver.io/ubuntu xenail/testing all Packages
Ign:12 https://repos.loraserver.io/ubuntu xenail/testing Translation-en_IN
Ign:14 https://repos.loraserver.io/ubuntu xenail/testing Translation-en
Ign:15 https://repos.loraserver.io/ubuntu xenail/testing amd64 DEP-11 Metadata
Ign:16 https://repos.loraserver.io/ubuntu xenail/testing DEP-11 64x64 Icons
Err:9 https://repos.loraserver.io/ubuntu xenail/testing amd64 Packages
  404  Not Found
Ign:10 https://repos.loraserver.io/ubuntu xenail/testing i386 Packages
Ign:11 https://repos.loraserver.io/ubuntu xenail/testing all Packages
Ign:12 https://repos.loraserver.io/ubuntu xenail/testing Translation-en_IN
Ign:14 https://repos.loraserver.io/ubuntu xenail/testing Translation-en
Ign:15 https://repos.loraserver.io/ubuntu xenail/testing amd64 DEP-11 Metadata
Ign:16 https://repos.loraserver.io/ubuntu xenail/testing DEP-11 64x64 Icons
Reading package lists... Done
[... cut ...]
E: Failed to fetch https://repos.loraserver.io/ubuntu/dists/xenial/InRelease  Unable to find expected entry 'testing/source/Sources' in Release file (Wrong sources.list entry or malformed file)
E: Failed to fetch https://repos.loraserver.io/ubuntu/dists/xenail/testing/binary-amd64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.
W: Target Sources (testing/source/Sources) is configured multiple times in /etc/apt/sources.list:55 and /etc/apt/sources.list:56
W: Target Sources (testing/source/Sources) is configured multiple times in /etc/apt/sources.list:55 and /etc/apt/sources.list:57

brocaar, 2017-08-02:
You have a typo in your configuration `xenail` must be `xenial`.

Also the log is hinting you about some duplicated config:

```
W: Target Sources (testing/source/Sources) is configured multiple times in /etc/apt/sources.list:55 and /etc/apt/sources.list:56
W: Target Sources (testing/source/Sources) is configured multiple times in /etc/apt/sources.list:55 and /etc/apt/sources.list:57
```

 As this is not a bug / issue related to the source-code of the LoRa Gateway Bridge component, please create a topic in the support forum in case you encounter further issues: https://forum.loraserver.io/

closed 2017-08-02 by brocaar