chirpstack/chirpstack-gateway-bridge #61: Gateway KeepAlive is not communicated over mqtt

Ankitjd18, 2017-10-13:
### Is this a bug or a feature request?
Bug (I guess)
### What did you expect?
To receive each keep alive or communication message through gateway bridge
### What happened?
For some time packets were not appearing on mqtt but communicating through gateway bridge
### What version are your using?
2.1.5
### How can your issue be reproduced?
No idea
### Could you share your log output?
Lora-gateway-bridge log output - it's working fine but no message when subscribing to mqtt for sometimes (recieving packets at random time not all the time)
`time="2017-10-13T09:10:41Z" level=info msg="gateway: sending udp packet to gateway" addr=14.102.105.90:56640 protocol_version=1 type=PullACK 
time="2017-10-13T09:10:51Z" level=info msg="gateway: received udp packet from gateway" addr=14.102.105.90:56640 protocol_version=1 type=PullData 
time="2017-10-13T09:10:51Z" level=info msg="gateway: sending udp packet to gateway" addr=14.102.105.90:56640 protocol_version=1 type=PullACK 
time="2017-10-13T09:11:01Z" level=info msg="gateway: received udp packet from gateway" addr=14.102.105.90:56640 protocol_version=1 type=PullData 
time="2017-10-13T09:11:01Z" level=info msg="gateway: sending udp packet to gateway" addr=14.102.105.90:56640 protocol_version=1 type=PullACK 
time="2017-10-13T09:11:11Z" level=info msg="gateway: received udp packet from gateway" addr=14.102.105.90:56640 protocol_version=1 type=PullData 
time="2017-10-13T09:11:11Z" level=info msg="gateway: sending udp packet to gateway" addr=14.102.105.90:56640 protocol_version=1 type=PullACK `

brocaar, 2017-10-13:
Are you referring to the [PULL_DATA](https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT#L279) packets received from the gateway? These are not forwarded over MQTT as the purpose is to keep firewalls open, not for communication with the network-server.

> That packet type is used by the gateway to poll data from the server.
>
>This data exchange is initialized by the gateway because it might be 
impossible for the server to send packets to the gateway if the gateway is 
behind a NAT.
>
> When the gateway initialize the exchange, the network route towards the 
server will open and will allow for packets to flow both directions.
The gateway must periodically send PULL_DATA packets to be sure the network 
route stays open for the server to be used at any time.

LoRa Gateway Bridge responds to these packets with a [PULL_ACK](https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT#L299).

Ankitjd18, 2017-10-13:
What are below mqtt topics for. I was looking for keep-alive messages sent by the gateway.
- gateway/[mac]/stats
- gateway/[mac]/rx
- gateway/[mac]/tx

Messages on these come at a fixed interval specified for the keep-alive message, but not come for few minutes randomly.

brocaar, 2017-10-14:
Please see: https://docs.loraserver.io/lora-gateway-bridge/use/data/.

Best topic would be the `.../stats` topic. You'll see data when your packet-forwarder is configured to send stats at every interval (e.g. every 30 seconds).

Ankitjd18, 2017-10-14:
I haven't listened anything on this topic from the beginning when subscribed with wild characters. Do I need to subscribe to this topic separately or something needs to be configured in the server?
Also, when I do not receive any message on the topics mentioned earlier data also stops coming. No device is able to communicate through the gateway.

brocaar, 2017-10-16:
I think https://forum.loraserver.io might be a better place to continue this discussion. I sense this has more to do with your gateway, networking and / or MQTT broker setup.

closed 2017-10-16 by brocaar

Ankitjd18, 2017-10-16:
Hey, I got confused in these. But, I have an issue that gateway does not respond for some time randomly, then it works after a restart. I tried to look for issues in logs but could not find anything.
Do you have any idea about what might be causing such issue or is this something I may be doing wrong?

brocaar, 2017-10-16:
As mentioned in my previous comment, I think https://forum.loraserver.io is a better place to find out what your issue is :-) Could you create a topic there, explaining your issue?

Ankitjd18, 2017-10-16:
Thanks for the help, I have created a topic - https://forum.loraserver.io/t/gateway-not-sending-data/240 to discuss this in the forum.