chirpstack/chirpstack-gateway-bridge #64: ca certificate format

martinadolfi, 2017-11-06:
### Is this a bug or a feature request?
Somewhere between a bug, a feature, and some info on the docs I guess ;)
### What did you expect?
Well, I don´t know what file format to use for the `--mqtt-ca-cert` option

### What happened?
I used `pem` and `der` format for the CA, but I got this error:
```bash
ERRO[0000] could not setup mqtt backend, retry in 2 seconds: Network Error : x509: certificate signed by unknown authority
```
Also, just to try, I passed any other file (not a certificate, i.e. the `start.sh` file suggested for kelink), and it gives the same error, it doesn't tell me there's an error on the certificate itself.

### What version are your using?
```bash
lora-gateway-bridge version 2.1.5
```
### How can your issue be reproduced?
Just try to use the --mqtt-ca-cert option not knowing the right certificate format ;)
```bash
/user/lora-gateway-bridge/bin/lora-gateway-bridge --mqtt-server ssl://dev-marting-3.exo.local:1883 --mqtt-ca-cert /user/lora-gateway-bridge/certnew.cer
```
### Could you share your log output?
```bash
/user/lora-gateway-bridge/bin/lora-gateway-bridge --mqtt-server ssl://dev-marting-3.exo.local:1883 --mqtt-ca-cert /user/lora-gateway-bridge/certnew.cer
INFO[0000] starting LoRa Gateway Bridge                  docs="https://docs.loraserver.io/lora-gateway-bridge/" version=2.1.5
INFO[0000] backend: connecting to mqtt broker            server="ssl://dev-marting-3.exo.local:1883"
ERRO[0000] could not setup mqtt backend, retry in 2 seconds: Network Error : x509: certificate signed by unknown authority
INFO[0002] backend: connecting to mqtt broker            server="ssl://dev-marting-3.exo.local:1883"
ERRO[0002] could not setup mqtt backend, retry in 2 seconds: Network Error : x509: certificate signed by unknown authority

```

brocaar, 2017-11-13:
This was implemented in https://github.com/brocaar/lora-gateway-bridge/pull/42. @minggi or @siscia could you give an answer on the above question?

siscia, 2017-11-13:
I believe that the problem is that the certificate is signed by an unknown autority.
#42 just add a certificate to the list of certificate to check.

Then I believe, don't quote me on this one, that is the MQTT library that check if the certificates is "correct"...

closed 2018-04-13 by brocaar