chirpstack/chirpstack-gateway-bridge #65: Support for OpenWrt

carlberg74, 2017-11-16:
If one of the designs is to have the lora-gateway-bridge running on the gateway itself 
and use MQTT instead of UDP.
It would be nice if the lora-gateway-bridge would support platforms like OpenWRT, 
as some gateway uses it.

Cheers
-Anders

brocaar, 2017-11-16:
The binary size can still be decreased a bit (using compile flags, no code changes required). Also it would be really easy to cross-compile to MIPS. See https://golang.org/doc/install/source#environment.

What would be needed / do you know the capabilities of these gateways you're mentioning? (e.g. available memory)

ivajloip, 2017-11-16:
BTW, if the size is the problem, I have been using [upx][1] for some time now to decrease the size to ~2MB for the ARM version.

[1]: https://en.wikipedia.org/wiki/UPX

carlberg74, 2017-11-16:
One of the gateways I have has only 10Mb RAM disk and 5Mb ROM, 
Where I have currently used 12% RAM.

But if you could build a lora-gateway-bridge for a ar71xx, that would be fantastic.
I would be happy to test it out.
Openwrt has support for mosquitto and I have found the necessary packages.

carlberg74, 2017-11-18:
I made an attempt to compile it for mips.. but I got some error.
(tested build with arm first and that worked)

/usr/local/go/src/github.com/brocaar/lora-gateway-bridge# GOOS=linux GOARCH=mips make build
Compiling source for linux mips
# github.com/brocaar/lora-gateway-bridge/vendor/github.com/jacobsa/crypto/cmac
vendor/github.com/jacobsa/crypto/cmac/hash.go:97:3: undefined: xorBlock
Makefile:8: recipe for target 'build' failed
make: *** [build] Error 2

Any ideas?

ivajloip, 2017-11-19:
Golang has the feature called build tags. This could be be used for a number of things, including for performance critical sections that could be re-written in the most efficient way. Unfortunately for the crypto lib there is no implementation for mips CPUs. Maybe you can ask the authors of the library if they can provide one.

brocaar, 2017-11-19:
@ivajloip would that be as easy as adding mips to the build tags, looking at the change you made last year? https://github.com/jacobsa/crypto/commit/181e0c043cc52efc103e0914a25270cb911d0565

ivajloip, 2017-11-20:
I think so, but this time I don't have the hardware to test it. I imagine the only thing to change is to add  *mips* and *mipsle* to [this line][1]. 

@carlberg74 if you will be able to test it on *mips* and *misple*, we can try to make a build and if all works well, make a pull request. 

[1]: https://github.com/jacobsa/crypto/blob/master/cmac/hash_32bit.go#L16

carlberg74, 2017-11-20:
@ivajloip added mips to some of the files and it compiled without any error.. thanks.

Found https://github.com/gomini/go-mips32 and where able to compile that as well..
But there no way the go runtime and the lora-gateway-bridge will fit on the device.
go-mips32/bin:
9.5Mb go
3.6Mb gofmt
11Mb linux_mips32 [dir]

any suggestions?

ivajloip, 2017-11-20:
1) the go version seems strange indeed. With go1.8 and up, there is official support for MIPS, even though I think it's considered experimental (still better than compiling it yourself). For me it was sufficient to write `GOARCH=mips make build` and it worked. Using the `-s` `-ldflags` removes some not really needed information (unless you want to use debuggers) and my executable is less than 6MB.
2) you don't need go and gofmt. You only need the executable itself. It should be a statically linked one, so no external dependencies are required. You just need to run it and see if it starts :)
3) `upx -9 build/lora-gateway-bridge` could even further reduce the size (<2MB).

carlberg74, 2017-11-21:
@ivajloip suddenly there is hope again. :) thanks. 
So I don't need gomini then, that's good news.

The reason I thought I needed go runtime, was that I got error: 'Illegal instruction' on a simple helloworld sample.
Continuing googling trying to find out what's wrong, I found that it could be due to lack of FSU emulation kernel support?
/sys/kernel/debug/mips/ don't have the file fpuemustats. aka I don't have FSU support.
Rather not rebuild the image.

Can I disable the need of FSU when building the go app or are there maybe another reason for this?

ivajloip, 2017-11-21:
Unfortunately here I can not help you... I suppose you mean FPU, because according to the [official release notes][1], FPU is needed (I don't know what is the exact difference between them). In any case, you need a kernel that supports it/can emulate it if you are to be able to run a program written in go.

[1]: https://golang.org/doc/go1.8#ports

brocaar, 2017-11-22:
FPU emulation is a config option in OpenWRT :-)https://forum.openwrt.org/viewtopic.php?id=25972

brocaar, 2017-12-01:
@ivajloip or @carlberg74 if you confirmed that the mips fix works in the https://github.com/jacobsa/crypto source, could you create a pull request for this? :-)

brocaar, 2017-12-08:
Go 1.10 beta has been announced: https://beta.golang.org/doc/go1.10

> On 32-bit MIPS systems, the new environment variable settings GOMIPS=hardfloat (the default) and GOMIPS=softfloat select whether to use hardware instructions or software emulation for floating-point computations.

This will make it much easier to deploy the LoRa Gateway Bridge onto a mips gateway I believe!

carlberg74, 2017-12-08:
I cant confirm its working yet, still struggling with getting the old mips image to compile having the kernel flag set with FPU.

These are the files in:
lora-gateway-bridge/vendor/github.com/jacobsa/crypto/cmac 
I have changed to get it jacobsa/crypto to compile.

hash_32bit.go:// +build 386 arm,!arm64, mips
hash_64bit.go:// +build amd64 arm64 ppc64 ppc64le s390x mips64 mips64le
hash_64bit.go:// +build 386 arm,!arm64, mips
hash.go:// +build 386 arm,!arm64, mips

Really good news about Go 1.10!

akaustel, 2018-01-31:
Did you get the kernel to compile with FPU?

carlberg74, 2018-01-31:
Yes, I did.
Haven't tested the image yet, but I hope soon..

akaustel, 2018-01-31:
I have enabled MIPS FPU in three separate places, but it still is not activated. Is there some secret involved you could share? I'm working on OpenWrt on an ar71xx.

akaustel, 2018-01-31:
Just got it working. The last option enabled was not clean compiled, now it is working.

brocaar, 2018-01-31:
Please note again that Go 1.10 will make  this much easier:
https://tip.golang.org/doc/go1.10#ports

Basically add the `GOMIPS=softfloat` env variable when compiling (if I understand correctly).

Once Go 1.10 is out, I'm happy to provide pre-compiled MIPS binaries with this set :-)

carlberg74, 2018-02-15:
Excellent news here also!
I can now run it on the gateway and I can see it starts and trying to connect to MQTT server etc!

lglenat, 2018-11-21:
Adding the MIPS architecture to goreleaser is easy but goreleaser does not have support for the GOMIPS env variable, so the binary can only be compiled with hardfloat (default), not softfloat.
Is it still worth adding to the goreleaser configuration?

I have the Gemtek Femto gateway here and it's running OpenWrt on a MT7620A. The kernel probably has hardfloat emulation because the hardfloat binary is working just fine (no need for softfloat), so I think the hardfloat binary is worth adding.

mrpackethead, 2019-03-10:
Whats the status of openwrt support?  I'd like to get the lora-gateway-bridge running on my RAK7242 gateway, which is running openwrt. 

root@6phr_por:/# cat /proc/cpuinfo
system type             : MediaTek MT7628AN ver:1 eco:2
machine                 : Mediatek MT7628AN evaluation board
processor               : 0
cpu model               : MIPS 24KEc V5.5
BogoMIPS                : 385.84
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16 dsp
shadow register sets    : 1
kscratch registers      : 0
package                 : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available

xueliu, 2019-07-31:
> Whats the status of openwrt support? I'd like to get the lora-gateway-bridge running on my RAK7242 gateway, which is running openwrt.
>
> root@6phr_por:/# cat /proc/cpuinfo
> system type : MediaTek [...]

Please have a try with my OpenWRT feed, where I try to integrate loraserver softwares into OpenWRT.

brocaar, 2019-11-23:
^^ that will include MIPS as build target for the next release :) This will be used to support the Dragino LG308 gateway (https://www.dragino.com/products/lora-lorawan-gateway/item/140-lg308.html).

commit b9749fcf (2019-11-23): Add mips build target (compressed with upx).

The binary for MIPS will be compressed using UPX since most MIPS devices
have limited memory available.

Closes #65.
closed 2019-11-23 by brocaar