chirpstack/chirpstack-gateway-bridge #66: lora-gateway-bridge didn`t  emit join-accept  message

MageekChiu, 2017-11-20:
I  tried to perform OTTA using this.
However it seems the gateway successfully sent join-request message to NS and AS agreed
here are some logs  , 

```
journalctl -u lora-app-server -u loraserver -u lora-gateway-bridge

Nov 20 20:18:22 lora-gateway-bridge[30219]: time="2017-11-20T20:18:22+08:00" level=info msg="gateway: received udp packet from gateway" addr=### protocol_version=2 type=P
Nov 20 20:18:22 lora-gateway-bridge[30219]: time="2017-11-20T20:18:22+08:00" level=info msg="gateway: rxpk packet received" addr=###data="ABERERERERERExERERERERE0DfFWPM0
Nov 20 20:18:22 lora-gateway-bridge[30219]: time="2017-11-20T20:18:22+08:00" level=info msg="backend: publishing packet" topic="gateway/0000000000000000/rx"
Nov 20 20:18:22 lora-gateway-bridge[30219]: time="2017-11-20T20:18:22+08:00" level=info msg="gateway: sending udp packet to gateway" addr=### protocol_version=2 type=Push
Nov 20 20:18:22 loraserver[30214]: time="2017-11-20T20:18:22+08:00" level=info msg="backend/gateway: rx packet received"
Nov 20 20:18:23 loraserver[30214]: time="2017-11-20T20:18:23+08:00" level=info msg="packet(s) collected" dev_eui=1111111111111113 gw_count=1 gw_macs=0000000000000000 mtype=JoinRequest
Nov 20 20:18:23 lora-app-server[30213]: time="2017-11-20T20:18:23+08:00" level=info msg="node updated" dev_eui=1111111111111113
Nov 20 20:18:23 lora-app-server[30213]: time="2017-11-20T20:18:23+08:00" level=info msg="join-request accepted" app_eui=1111111111111111 application_name=ludeng dev_addr=073703f6 dev_eu
Nov 20 20:18:23 lora-app-server[30213]: time="2017-11-20T20:18:23+08:00" level=info msg="handler/mqtt: publishing join notification" topic="application/2/node/1111111111111113/join"
Nov 20 20:18:23 loraserver[30214]: time="2017-11-20T20:18:23+08:00" level=info msg="node-session saved" dev_addr=073703f6 dev_eui=1111111111111113
Nov 20 20:18:23 loraserver[30214]: time="2017-11-20T20:18:23+08:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/0000000000000000/tx"

tcpdump udp port 1700
20:18:22.851293 IP gateway >server.1700: UDP, length 197
20:18:22.851843 IP server > gateway: UDP, length 4
```
according to tcpdump there is only ACK message , no  join accept message。

brocaar, 2017-11-20:
LoRa Gateway Bridge doesn't know anything about the LoRaWAN protocol, so it doesn't know about join-accepts. It only forwards LoRa payloads (which so happens to be a join-request / join-accept). I don't understand your question related to the ACK message, but I think you are referring to the UDP ACK, see also [PROTOCOL.txt](https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT).

MageekChiu, 2017-11-20:
Sorry for the confusion.
ACK means PUSH_ACK in PROTOCOL.txt, it`s a 4-bytes packet .
My problem is that  the  procedure of OTAA is like this:

node --(join request)--> gateway -> LoRa Gateway Bridge(send PUSH_ACK to gateway immediately and then publish MQTT message) -> loraserver -> lora-app-server--(join-accept)--> loraserver -> MQTT -> LoRa Gateway Bridge -> gateway.

so the gateway should receive 2 packets if everything went right,one is  PUSH_ACK , the other is join-accept, however , according to tcpdump there is only ACK message , no join-accept message。

brocaar, 2017-11-22:
Are you running the LoRa Gateway Bridge on the gateway itself? If not, could you give that a try?

MageekChiu, 2017-11-23:
thank you for your time, I`ll try running the LoRa Gateway Bridge on the gateway

brocaar, 2017-11-23:
Perfect, please let us know if that solved your issue (I expect it will), so we can close the issue afterwards.

MageekChiu, 2017-11-23:
After reading  the codes and logs, I find out that the LoRa Gateway Bridge didn`t subscribe to the tx channel, and thus it wasn`t able to get data published by the Loraserver, let along sending data to gateway. 
And I found that only by handling PULL_DATA first will the LoRa Gateway Bridge subscribe to the tx channel, and the gateway did`t send PULL_DATA first , that`s the problem, and I fixed it.
the PROTOCOL.txt did tell us that:
```
The gateway must periodically send PULL_DATA packets to be sure the network 
route stays open for the server to be used at any time.
```
I should have read it carefully, my bad.
thanks again for your time

closed 2017-11-23 by MageekChiu

ahmadkarim, 2018-02-05:
How did you handle the PULL_DATA, did you change something in the Lora-gateway-bridge or you wrote some script for that. I have a setup like this:

concentrator--(spi)-->Packet forwarder--(udp)-->Lora-gateway-bridge--(mqtt/pub)-->Mosquitto--(mqtt/sub)-->myScript"

 i want to transmit some data to my node, so where should i handle the PULL_DATA ?

gclairec, 2018-02-06:
I would also like to know where to handle the PULL_DATA

brocaar, 2018-02-06:
The whole idea of the LoRa Gateway Bridge is that you don't need to implement the UDP protocol yourself / deal with PULL_DATA. Simply use the MQTT topics as documented here: https://docs.loraserver.io/lora-gateway-bridge/use/data/. LoRa Gateway Bridge will handle the UDP protocol specifics for your.

gclairec, 2018-02-06:
Do you have any documentation regarding the hosted version for loraserver?

brocaar, 2018-02-06:
If you mean the server hosted by Kyrio, the documentation hosted at https://docs.loraserver.io/ applies to Kyrio too. For other questions it is better to contact Kyrio directly :-)