chirpstack/chirpstack-gateway-bridge #67: lora-gateway-bridge doesn`t send data to gateway

MageekChiu, 2017-11-22:
I`m using [this](https://docs.loraserver.io/lora-app-server/integrate/data/)

`application/[applicationID]/node/[devEUI]/tx `  to send data to node

the mqtt log is:
```
application/2/node/1111111111111113/tx {"reference":"abcd1234","confirmed":false,"fPort":10,"data":"YWRhZGFkYWQ="}
gateway/0000000000000000/tx {"txInfo":{"mac":"0000000000000000","immediately":true,"timestamp":0,"frequency":869525000,"power":14,"dataRate":{"modulation":"LORA","spreadFactor":9,"bandwidth":125},"codeRate":"4/5","iPol":null},"phyPayload":"YA/odgYAAAAKtkaC+BvYdfJx7Z1K"}

```
the lora log  is:

```
Nov 21 18:50:24 iZbp11ivbiy94hpjfdsqs7Z lora-app-server[15372]: time="2017-11-21T18:50:24+08:00" level=info msg="handler/mqtt: data-down payload received" topic="application/2/node/1111111111111113/tx"
Nov 21 18:50:24 iZbp11ivbiy94hpjfdsqs7Z loraserver[15378]: time="2017-11-21T18:50:24+08:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/0000000000000000/tx"
Nov 21 18:50:24 iZbp11ivbiy94hpjfdsqs7Z loraserver[15378]: time="2017-11-21T18:50:24+08:00" level=info msg="node-session saved" dev_addr=0676e80f dev_eui=1111111111111113
```
and `tcpdump udp port 1700` shows nothing , which means lora-gateway-bridge isn`t able send data to gateway .

this looks like the same issue as[ I described before](https://github.com/brocaar/lora-gateway-bridge/issues/66), 

To summarise，the log "gateway: sending udp packet to gateway"  in [sendPackets](https://github.com/brocaar/lora-gateway-bridge/blob/master/gateway/backend.go) only shows up whit type type=PushACK, which mean  lora-gateway-bridge couldn`t take the initiative to send data to gateway   

please help . thanks a lot.

brocaar, 2017-11-22:
I don't experience any issues myself, I think it could be due to your environment (NAT, firewall, settings). I'll close this issue as I think it is the same issue as https://github.com/brocaar/lora-gateway-bridge/issues/66. If not, we'll reopen this one.

closed 2017-11-22 by brocaar

vishalchoudhary0011, 2025-07-18:
https://github.com/Lora-net/sx1302_hal/issues/143#issue-3240194999

#[brocaar](https://github.com/brocaar) please give me guidance for solving this issue