chirpstack/chirpstack-gateway-bridge #71: DebugLevel

alexanderplanar, 2017-12-15:
### Is this a bug or a feature request?
Bug, I guess

### What did you expect?
I expected to see debug messages

### What happened?
No debug messages reported

### What version are your using?
2.2.0

### How can your issue be reproduced?
$ lora-gateway-bridge --log-level=5 ....

### Could you share your log output?
it is not needed, the log contains only "INFO" messages

I try to fix the problem with the ACK packets. The gateway rejects the packets with JIT error 1 and 2 ("wrong timestamp too much in advance" and "too late to send")
I installed gateway bridge on the gateway host and try to run it in debug mode.
I modified /etc/default/lora-gateway-bridge file and changed the LOG_LEVEL value to 5.
But it does not help. I tried to run the application directly (as an executable, but not as a service), providing the same parameters via the command line  including --log-level=5, but it does not help too.
Please, help to fix the problem with the debug mode. Thanks!

brocaar, 2017-12-15:
Maybe logging could be improved by adding (more) debug-level logs. If you would be able to contribute on this by creating a pull-request with more logging, that would be great and much appreciated!

brocaar, 2018-01-17:
Actually, I don't think there is more info that can be displayed. All the traffic from / to the packet-forwarder is already debugged. The "wrong timestamp too much in advance" error is exactly the error that is returned by the packet-forwarder. I could print it as a debug logline, but that wouldn't change the information exposed ;-)

closed 2018-05-10 by brocaar