chirpstack/chirpstack-gateway-bridge #75: Tektelic pico gateway doesn't transmit packet from lora-gateway-bridge with "brd" and "ant" fields present.

vvvua, 2018-01-17:
BUG:
With OTAA using Tektelic pico gateway doesn't transmit the answer from lora-gateway-bridge to lorawan device if "brd" and "ant" fields are present.
That packet failed to be transmitted:
{
“txpk”: {
“imme”: false,
“tmst”: 2644304043,
“freq”: 868.5,
“rfch”: 0,
“powe”: 14,
“modu”: “LORA”,
“datr”: “SF7BW125”,
“codr”: “4/5”,
“ipol”: true,
“size”: 17,
“data”: “IOGbxyNJucfqHxjNJ4H0Qr0=”,
“brd”: 0,
“ant”: 0
}
}

This one is ok:

{
  "txpk": {
    "imme": false,
    "tmst": 311856371,
    "freq": 868.1,
    "rfch": 0,
    "powe": 14,
    "modu": "LORA",
    "datr": "SF7BW125",
    "codr": "4/5",
    "ipol": true,
    "size": 17,
    "data": "ICHJHdB6pwYQ7QkPiFNX\/nk="
  }
}

lora-gateway-bridge version 2.2.0

Diff file for tests only. Not checked properly!

[backend.diff.txt](https://github.com/brocaar/lora-gateway-bridge/files/1638644/backend.diff.txt)

brocaar, 2018-01-17:
Hmm, that is a pain, since these fields are used by some Kerlink gateways that have multiple boards and / or antennas (they use a slightly adapted packet-forwarder version).

It seems that Tektelic implemented their own packet-forwarder as https://github.com/lora-net/packet_forwarder works fine with these fields while they are unknown to the single board / antenna version. Preferrably the Tektelic firmware would just ignore these fields too...

Do you have a technical contact at Tektelic, would be happy to work with them to see how this can be resolved in the best way.

vvvua, 2018-01-17:
Tektelic uses FreeRTOS in that gateway (by documentation). So packet forwarder has to be different.
I'll try to find contacts of Tektelic.

brocaar, 2018-01-23:
@vvvua I've been in contact with Tektelic and they have a firmware update. They also would like you to contact them so you can get the update: 

> We have created a new version of our software which ignores these fields and also sends the ‘stat’ message. Can you refer the person who had this question directly to me to get the software update?

Could you send me an e-mail? (info [at] brocaar [dot] com). Then I'll forward you the contact details to get in touch with Tektelic so that you can get / test the new firmware.

vvvua, 2018-01-28:
Fixed in new firmware (semtech-v1.1.2.bin) of Tektelic pico gateway.
Thank you, brocaar!

closed 2018-01-28 by vvvua