chirpstack/chirpstack-gateway-bridge #77: Sending decrypted LoRa Frame Payload to the Fiware's JSON/MQTT IOT agent

ktsakos, 2018-01-28:
I have a gateway which catches LoRa packets(decrypted with AES-128 and encoded as base64) and converts them to UDP. In addition, i execute a service on my gateway (lora-gateway-bridge: https://github.com/brocaar/lora-gateway-bridge) which abstracts the UDP protocol and converts my packet to MQTT. On cloud,the packets are published as they are to the mosquitto MQTT broker in JSON format which includes the gateway's metadata and the phypayload as key value pairs. I also run the Fiware's JSON/MQTT IOT agent service and the Context Broker. What should i do in order to send the measures from the mosquitto to the IOT agent in order to be compatible? For example i want to make a post request to iotagent service to send the measure of the temperature and humidity....these are published until now to the mosquitto as

gateway/1dee18c14948a955/rx {"rxInfo":{"mac":"1dee18c14948a955","time":"0001-01-01T00:00:00Z","timestamp":38099540,"frequency":868500000,"channel":2,"rfChain":1,"crcStatus":1,"codeRate":"4/5","rssi":-28,"loRaSNR":9,"size":20,"dataRate":{"modulation":"LORA","spreadFactor":9,"bandwidth":125}},"phyPayload":"gFY0EjoAAgABCCpbg+AkKM1gHnQ="}

Essentially i want to send to my IOT agent the decrypted frame payload that the phypayload includes,which is 006700ED016866 HEX.This means that the temperature is 00ED(HEX) and the humidity 66(HEX) according the cayenne low power payload format. Then i had to post to the IOT agent the temperature t:23,7 and the humidity h:54

brocaar, 2018-01-29:
To be sure, is `gFY0EjoAAgABCCpbg+AkKM1gHnQ="` a LoRa or LoRaWAN frame? You mention the data is *decrypted with AES-128*. Do you mean it is **not** encrypted?

In case it is a non-LoRaWAN payload but just a blob that holds your data, then the only thing needed to do is to decode it according to the Cayenne LPP encoding and forward this data. But this functionality you have to develop yourself somewhere in your pipeline.

brocaar, 2018-01-29:
Please keep the discussion at: https://forum.loraserver.io/t/lora-app-server-compatibility-integration-with-fiware-and-others/167.

closed 2018-01-29 by brocaar

ktsakos, 2018-01-29:
It's a LoRa

brocaar, 2018-01-29:
In that case you need to implement an application which decodes your `phyPayload` into the sensor values.

ktsakos, 2018-01-29:
Is this operation supported from lora-app-server? I think that it supports the decryption of the Cayenne low power payload,doesn't it?

brocaar, 2018-01-29:
Yes it does, but only if your device operates in LoRaWAN.

ktsakos, 2018-01-29:
Yes my packets from the device are transmitted as LoRa with Cayenne format of their frame payload!So should i use all your lora-project,not only the lora-gateway-bridge and i will publish from the lora-app-server decrypted attributes with their values as JSON to the mosquitto?