chirpstack/chirpstack-gateway-bridge #90: ApplyConfiguration panic

gq-tang, 2018-08-28:
### Out of index panic occurs when the gateway channel in loraserver is set to be greater than 8.

<pre>
if multiSFCounter >= channelCount {
					return gc, errors.New("gateway: exceeded maximum number of multi-sf channels")
				}
</pre>

brocaar, 2018-08-28:
The error:

`errors.New("gateway: exceeded maximum number of multi-sf channels")`

is intentional as the sx1301 / sx1308 on your gateway does not support > 8 multi-spreadingfactor channels.

gq-tang, 2018-08-28:
I see, but if the number of channels set on loraserver is >8, gateway-bridge will panic instead of returning an error.
> should be changed to >= in [source code](https://github.com/brocaar/lora-gateway-bridge/blob/master/internal/gateway/semtech/pf_config.go#L247)

brocaar, 2018-08-28:
Ah, good catch! Would you like to make a pull-request to change this?

brocaar, 2018-08-31:
Closing as the pull-request fixing this has been merged.

closed 2018-08-31 by brocaar