chirpstack/chirpstack-gateway-bridge #104: json rxpk.brd parsing value

gillespilloudwyres, 2019-03-04:
### Is this a bug or a feature request?
maybe a bug

### What did you expect?
On my lora-gateway-bridge the packets are not transmitted to the loraserver due to a json parsing error :

[image]

on the JSON, the brd value is 261 (correct uint8 value)

Have you an idea about the root cause error ?

gillespilloudwyres, 2019-03-05:
I can't see the brd field in the https://github.com/Lora-net/packet_forwarder/blob/experimental/PROTOCOL.TXT i will check with my packet forwarder provider if the value is correct and if values upper than 255 can will be provide in the future.

brocaar, 2019-03-05:
The `brd` is used by for example the iBTS gateway as it can contain up to 4 boards (each serving 16 channels). I could change from `uint8` to a type able to store larger numbers, but it would be good to validate what `brd: 261` means :-)

gillespilloudwyres, 2019-03-05:
So,

on the https://github.com/brocaar/loraserver/blob/master/api/gw/gw.pb.go#L558 it's defined as an uint32.

From kerlink collegues :

On PROTOCOL.TXT and the protocol GWMP 1.5, Semtech has defined a brd field as :
brd  | number | Concentrator board used for RX (unsigned integer)

There is no size defined. the lns must transmit the value without change (blackbox for the LNS).

Kerlink use this field to transmit more information on it :

Bits | Name | Description
-- | -- | --
31-24 |   | Unused
23-20 | Zone | Antenna zone (refers to the system)
19-16 | Antenna | Antenna identifier (refers to the system)
15-12 | Board | Board identifier (refers to the system)
11-8 | Transceiver | Transceiver identifier (refers to the board)
7-4 | Demodulator | Demodulator identifier (refers to the board)
3-0 | Channel | Channel identifier (refers to the demodulator)

is it ok for you to transform the field to uint32 (I can prepare a PR) ?

brocaar, 2019-03-05:
Yes, please do :-)

commit 82adc02b (2019-03-07): Change Brd from uint8 to uint32.

Some packet-forwarders encode more information in this field and
therefore require require an uint32 type instead of uint8.

Closes #104 and #105. Thanks @gillespilloudwyres.
closed 2019-03-07 by brocaar