chirpstack/chirpstack-gateway-bridge #112: lora-gateway-bridge bug

wateras, 2019-04-09:
- 1. ### Is this a bug or a feature request?
- 2. I used the lora-gateway-bridge component to deploy on my arm a7 platform. I found that when the server sends a message to the bridge at the timing (20s), there will be no downlink data received. Basically, the number of running times is about 160 times. When you subscribe to the gateway/aa00551100110111/tx and gateway/aa00551100110111/ack messages with the mqtt client software, you can only see the tx topic message, but there is no ack message. The normal situation should be that the tx and ack times are the same. What is the reason for this? When this happens, the upstream data sent by the gateway to the server is not affected.
- 3. ### What did you expect?
- 4. I hope that you can help solve this problem.
- 5. ### What happened?
- 6. 
- 7. ### What version are your using?
- 8. lora-gateway-bridge_2.7.1_linux_armv7.tar.gz
- 9. ### How can your issue be reproduced?
- 10. 
- 11. ### Could you share your log output?
- 12. [image]

brocaar, 2019-04-11:
The ack is sent by the Semtech packet-forwarder, LoRa Gateway Bridge only forwards this ACK. So when the packet-forwarder does not send ACKs, LoRa Gateway Bridge won't either.

Can you confirm if the Semtech packet-forwarder sends ACKs? You will find some helpful commands here: https://www.loraserver.io/guides/troubleshooting/gateway/.

wateras, 2019-04-12:
> The ack is sent by the Semtech packet-forwarder, LoRa Gateway Bridge only forwards this ACK. So when the packet-forwarder does not send ACKs, LoRa Gateway Bridge won't either.
>
> Can you confirm if [...]

I did not see the received downlink data log from the log of the gateway program. Did the gateway bridge not receive the downlink data pushed by the server at all? At present, our server architecture is a gateway bridge deployed on the server. I also deployed a bridge on the arm a7 platform. I found that the communication also has problems. The phenomenon is that the server sometimes does not see the uplink data, but the mqtt client software does. You can subscribe to the gateway/mac/rx message.

closed 2019-06-24 by brocaar