chirpstack/chirpstack-gateway-bridge #114: Allow setting MessageChannelDepth

tomasol, 2019-04-25:
### Is this a bug or a feature request?
Feature request
### What did you expect?
I would like to set paho's MessageChannelDepth via config file.
Currently I believe one can enquee up to 100 messages to be published when connection is down, based on [this](https://github.com/eclipse/paho.mqtt.golang/blob/master/options.go#L112).
Please allow changing this via the [setter](https://github.com/eclipse/paho.mqtt.golang/blob/master/options.go#L330).

blocknonip, 2019-05-01:
I didn't realize this feature already existed and was going to request this. +1, please make this configurable in the gateway bridge configuration.

brocaar, 2019-10-21:
I'm closing this issue as this doesn't behave as one would expect, see the above link.

closed 2019-10-21 by brocaar