chirpstack/chirpstack-gateway-bridge #116: mqtt connection error, error=EOF

tankienleong, 2019-05-02:
### Is this a bug or a feature request?
a bug

### What did you expect?
lora gateway bridge send mqtt data to Google Cloud IoT Core

### What happened?

### What version are your using?
lora gateway bridge v2.7.1

### How can your issue be reproduced?

### Could you share your log output?
[image]

brocaar, 2019-05-02:
Without any details on how to re-produce this (like your LoRa Gateway Bridge configuration, how you have setup your device in the GCP IoT Core, ...) there is not much I can help you with.

tankienleong, 2019-05-02:
Hi @brocaar , I follow the tutorial on [](https://www.loraserver.io/guides/google-cloud-platform/)
The lora-gateway-bridge is running on a raspberry pi.
Below is the content of lora-gateway-bridge.toml
[general]
# debug=5, info=4, warning=3, error=2, fatal=1, panic=0
log_level = 4

# Configuration which relates to the packet-forwarder.
[packet_forwarder]
udp_bind = "0.0.0.0:1700"

skip_crc_check = false

  # # Managed packet-forwarder configuration.
  # #
  # # By configuring one or multiple managed packet-forwarder sections, the
  # # LoRa Gateway Bridge updates the configuration when the backend receives
  # # a configuration change, after which it will restart the packet-forwarder.
  # [[packet_forwarder.configuration]]
  # # Gateway MAC.
  # #
  # # The LoRa Gateway Bridge will only apply the configuration updates for this
  # # gateway MAC.
  # mac="0102030405060708"

  # # Base configuration file.
  # #
  # # This file will be used as base-configuration and will not be overwritten on
  # # a configuration update. This file needs to exist and contains the base
  # # configuration and vendor specific
  # base_file="/etc/lora-packet-forwarder/global_conf.json"

  # # Output configuration file.
  # #
  # # This will be the final configuration for the packet-forwarder, containing
  # # a merged version of the base configuration + the requested configuration
  # # update.
  # # Warning: this file will be overwritten on a configuration update!
  # output_file="/etc/lora-packet-forwarder/local_conf.json"

  # # Restart command.
  # #
  # # This command is issued by the LoRa Gateway Bridge on a configuration
  # # change. Make sure the LoRa Gateway Bridge process has sufficient
  # # permissions to execute this command.
  # restart_command="/etc/init.d/lora-packet-forwarder restart"

# Configuration for the MQTT backend.
[backend.mqtt]
uplink_topic_template="gateway/{{ .MAC }}/rx"
downlink_topic_template="gateway/{{ .MAC }}/tx"
stats_topic_template="gateway/{{ .MAC }}/stats"
ack_topic_template="gateway/{{ .MAC }}/ack"
config_topic_template="gateway/{{ .MAC }}/config"

# Payload marshaler.
marshaler="protobuf"

# MQTT authentication.
[backend.mqtt.auth]  
type="gcp_cloud_iot_core"

# Generic MQTT authentication.
#[backend.mqtt.auth.generic]
#server="tcp://127.0.0.1:1883"

# Connect with the given username (optional)
username=""

# Connect with the given password (optional)
password=""

# Quality of service level
qos=0

# Clean session
clean_session=true

# Client ID
client_id=""

# CA certificate file (optional)
ca_cert=""

# mqtt TLS certificate file (optional)
tls_cert=""

# mqtt TLS key file (optional)
tls_key=""

# Maximum interval that will be waited between reconnection attempts when connection is lost.
# Valid units are 'ms', 's', 'm', 'h'. Note that these values can be combined, e.g. '24h30m15s'.
max_reconnect_interval="10m0s"

# Google Cloud Platform Cloud IoT Core authentication.
[backend.mqtt.auth.gcp_cloud_iot_core]
# MQTT server.
server="ssl://mqtt.googleapis.com:8883"

# Google Cloud IoT Core Device id.
device_id="gw-utem-000001"

# Google Cloud project id.
project_id="utemloraserver"

# Google Cloud region.
cloud_region="asia-east1"

# Google Cloud IoT registry id.
registry_id="utem-gateways"

# JWT token expiration time.
#jwt_expiration="24h0m0s"

jwt_key_file="/home/risinghf/private-key.pem"

Below are the screenshots of configuration of Google Cloud IoT core:
[image]
[image]
[image]

Please let me know if you required more information. 
Thank you very much.

brocaar, 2019-05-03:
It disconnects since the Device ID and gateway ID (with `gw-` prefix) don't match. The device ID in GCP Cloud IoT Core must be `gw-GATEWAYID` where `GATEWAYID` is the 64bit gateway ID (`gw-b8......73`).

closed 2019-05-03 by brocaar