chirpstack/chirpstack-gateway-bridge #117: LWT message support

alexbondarchuk, 2019-05-21:
### Is this a bug or a feature request?
 Feature request
### What d you expect?
Add following possibilities:	
```      
        WillEnabled             bool
	WillTopic               string
	WillPayload             []byte
	WillQos                 byte
	WillRetained            bool
```
to lora-gateway-bridge config file for enable LWT feature.
This options supported by [paho.mqtt.golang client](https://github.com/eclipse/paho.mqtt.golang/blob/master/options.go).

4refr0nt, 2019-06-24:
This feature may be very useful for gateways monitoring

brocaar, 2019-12-10:
I did see @sa-wilson  his pull-request and I have been thinking about this feature for a while but I'm not sure if the ChirpStack Gateway Bridge should support this. As I understand, a LWT payload is usually used in the following way:

* A MQTT device connects and publishes a retained message (let's say `online`) to e.g. `device/123/status` 
* The same device then also publishes a retained LWT message (let's say `offline`) to the same topic `device/123/status`

When an other MQTT client subscribes to `device/123/status`, it will receive `online` when the device is online, or `offline` when it disconnected.

The "issue" with the ChirpStack Gateway Bridge: when it starts, it has no knowledge which gateway(s) it is serving. Also, it is possible to connect multiple Packet Forwarder instances to a single Gateway Bridge. Ideally you would have a `gateway/0102030405060708/status` or `gateway/0102030405060708/event/status` LWT topic, but this is not possible as far as I know.

I would really like to avoid to add this as a fully generic function where the end-user in charge of all the parameters, including the configuration of the (potential binary) payload. This makes it really hard if not impossible to later integrate this with the other components.

I think a better way of gateway monitoring is to use the Prometheus metrics that the Gateway Bridge provides. Also, since most Packet Forwarder instances are configured to post stats every 30 seconds, you will see a `gateway/GATEWAYID/event/stats` message. This "heartbeat" can also be used for monitoring.

See also #85

brocaar, 2021-06-29:
Please see https://www.chirpstack.io/gateway-bridge/changelog/#v3110 (Connection state messages).

closed 2021-06-29 by brocaar