chirpstack/chirpstack-gateway-bridge #120: Improve 'apply gateway-configuration error' error on re-configuration

brocaar, 2019-05-27:
### Is this a bug or a feature request?

Improvement

### What did you expect?

A better error message when the gateway is not configured for automatic re-configuration. Currently it logs "apply gateway-configuration error: gateway does not exist".

### What happened?

The error log is confusing.

### What version are your using?

Reproducible in latest version.

### How can your issue be reproduced?

Send a `.../command/config` message when no `[[backend.semtech_udp.configuration]]` section is present in the `lora-gateway-bridge.toml` file.

### Could you share your log output?

> may 24 15:39:31 localhost lora-gateway-bridge[22056]: time="2019-05-24T15:39:31-03:00" level=error msg="apply gateway-configuration error" error="gateway does not exist"

Aaron-Yue, 2019-06-17:
Hi, @brocaar, I'm new to Lora, I now have this error information  
`level=error msg="apply gateway-configuration error" error="gateway does not exist"`
in the bridge log. I'm confused whether the gateway can be used? What should I do to solve this error?
More log like below:
`Jun 17 16:45:31 VM-0-35-ubuntu lora-gateway-bridge[15191]: time="2019-06-17T16:45:31+08:00" level=info msg="integration/mqtt: publishing event" event=stats qos=0 topic=gateway/b827ebfffeb603fd/event/stats
Jun 17 16:45:31 VM-0-35-ubuntu lora-gateway-bridge[15191]: time="2019-06-17T16:45:31+08:00" level=info msg="integration/mqtt: gateway configuration received" topic=gateway/b827ebfffeb603fd/command/config
Jun 17 16:45:31 VM-0-35-ubuntu lora-gateway-bridge[15191]: time="2019-06-17T16:45:31+08:00" level=error msg="apply gateway-configuration error" error="gateway does not exist"`

brocaar, 2019-07-24:
This feature will be deprecated and the error has been removed.

closed 2019-07-24 by brocaar