chirpstack/chirpstack-gateway-bridge #128: Packet-Forwarder does not start on startup (Multitech gateway)

EtienneBourdeau, 2019-08-01:
Hello !

I have followed the tutorial of this page to install the packet forwarder on my Multitech gateway :
https://www.loraserver.io/lora-gateway-bridge/gateway/multitech/

But, while it worked perfectly on older gateways (same model, just older mlinux versions), it happens that on mlinux 5.0.0, after adding the packet forwarder to defaults with update-rc.d and rebooting, an error message appears on startup : 

```
Found MTAC-LORA-H-868 with MTAC-LORA-1.5 hardware at ap1
Starting lora-packet-forwarder: renice: getpriority: No such process
OK
```

Is there a way to fix this ?
By advance, thank you very much !

Kind regards,
Etienne

brocaar, 2019-08-01:
Are you sure the logged error is from starting the packet-forwarder (I've seen sometimes that when processes are logging async, that lines can get mixed).

On my Conduit I have installed according to the instructions you mentioned. After a reboot, I do see that the forwarder is started.

EtienneBourdeau, 2019-08-02:
Thank you for your answer.

I will be a little more precise : 
This morning (Friday August 2nd, ~09h00 AM GMT+1), I have flashed once again the 5.0.0 firmware on my mlinux Gateway (model : MTCDT-LEU1-246L-EU-GB), so I could go all over again from the beginning with a "clean" gateway

Here are the steps I have followed from here :

- PPP configuration (my gateway functions by GSM)
- Installing the packet forwarder through the instructions of your tutorial here : https://www.loraserver.io/lora-gateway-bridge/gateway/multitech/
(Gateway Bridge not installed yet)
- reboot with USB serial plugged so I could see booting steps.
- I had this result, which seemed fine

https://image.noelshack.com/fichiers/2019/31/5/1564730561-capt1.png

- Logged in on the gateway
- Tried "ps aux"
- No packet forwarder in the list (see below)

https://image.noelshack.com/fichiers/2019/31/5/1564730561-capt2.png

- Tried to configure ppp on boot (with update-rc.d /etc/ppp defaults)
- reboot once again the gateway
- I had this result, with the renice problem

https://www.noelshack.com/2019-31-5-1564730561-capt3.png

- I have removed from launch the Multitech Packet Forwarder (update-rc.d -f lora-packet-forwarder remove), then rebooted, but same result.
- I have removed from launch your packet forwarder (update-rc.d -f lora-packet-forwarder-ap1 remove), then re-added it while gaving it the same priority the Multitech Packet Forwarder had (update-rc.d lora-packet-forwarder-ap1 defaults 80 30)/
- Now, I am back to the 1st problem : The renice priority problem seems solved, but I can't see the packet forwarder with ps aux, and it is not detected by the LoRaServer I installed, so it really seem not to run.

Thank you very much for reading this and for caring about this problem,
Kind regards,

Etienne

brocaar, 2019-08-02:
I have received a similar report:

> The semtech packet forwarder does not start up if the MT Conduit's cellular interface is enabled and it does not get a signal from the carrier.

One option would be to look into adding Monit as a service monitoring daemon. With the init script it either starts or fails but it will not re-retry. Using Monit (which I'm also using for the LoRa Gateway OS), it will retry if it detects that the process has failed.

closed 2020-10-05 by brocaar