chirpstack/chirpstack-gateway-bridge #147: New v3.5.0 MQTT Servers cannot be configured by environmental variable

blocknonip, 2019-11-27:
I've tried every variation I can think of using dots/double underscores and quotes, including:
```
- INTEGRATION__MQTT__AUTH__GENERIC__SERVERS=["tcp://mqtt:1883"]
```
```
- INTEGRATION__MQTT__AUTH__GENERIC__SERVERS="[tcp://mqtt:1883]"
```
```
- INTEGRATION__MQTT__AUTH__GENERIC__SERVERS=[tcp://mqtt:1883]
```
```
- INTEGRATION.MQTT.AUTH.GENERIC.SERVERS=[tcp://mqtt:1883]
```
...

And it always results with:
```
chirpstack-gateway-bridge_1      | time="2019-11-27T14:39:57Z" level=error msg="integration/mqtt: connection error" error="No servers defined to connect to"
```

Is something like this needed? https://github.com/brocaar/chirpstack-network-server/commit/003202e82b79a3129b50990b68395d2e6deb7ad5

brocaar, 2019-11-27:
Hi @blocknonip, yes I think you're right that the same fix is needed for slice configuration with env. variables when configuring multiple servers. Will create a fix for this!

brocaar, 2019-11-27:
Actually, there is no change needed. For the NS this was needed as it was not a slice of strings but a slice of structs. In this case just comma separate the list of servers :slightly_smiling_face: 

```
$ env INTEGRATION__MQTT__AUTH__GENERIC__SERVERS='tcp://127.0.0.1:1883,tcp://127.0.0.2:1883,tcp://127.0.0.3:1883' ./build/chirpstack-gateway-bridge configfile |grep servers -B 1 -A 10

    [integration.mqtt.auth.generic]
    # MQTT servers.
    #
    # Configure one or multiple MQTT server to connect to. Each item must be in
    # the following format: scheme://host:port where scheme is tcp, ssl or ws.
    servers=[
      "tcp://127.0.0.1:1883",
      "tcp://127.0.0.2:1883",
      "tcp://127.0.0.3:1883",
    ]

    # Connect with the given username (optional)
    username=""

    # Connect with the given password (optional)
    password=""
```

blocknonip, 2019-11-27:
This is very strange, especially in the context of a single server. Using:

```
- INTEGRATION__MQTT__AUTH__GENERIC__SERVERS="tcp://mqtt:1883"
```

yields the above `No servers defined to connect to` error message.

Using:

```
- INTEGRATION__MQTT__AUTH__GENERIC__SERVERS="tcp://mqtt:1883,"
```

yields:

```
chirpstack-gateway-bridge_1      | time="2019-11-27T17:38:18Z" level=error msg="integration/mqtt: connection error" error="Network Error : dial tcp: address \": missing port in address"
```

However, using:

```
- INTEGRATION__MQTT__AUTH__GENERIC__SERVERS=tcp://mqtt:1883
```

connects. The difference between (1) and (3) in the context of docker-compose is not obvious to me. Either way, it's all about as clear as mud, but I have a working version now. I will send a chirpstack-docker PR.

brocaar, 2019-11-28:
Hmm, that is odd. From the CLI:

```bash
$ env INTEGRATION__MQTT__AUTH__GENERIC__SERVERS="tcp://127.0.0.2:1883" ./build/chirpstack-gateway-bridge configfile |grep servers -B 1 -A 10

    [integration.mqtt.auth.generic]
    # MQTT servers.
    #
    # Configure one or multiple MQTT server to connect to. Each item must be in
    # the following format: scheme://host:port where scheme is tcp, ssl or ws.
    servers=[
      "tcp://127.0.0.2:1883",
    ]

    # Connect with the given username (optional)
    username=""

    # Connect with the given password (optional)
    password=""

    # Quality of service level
```

blocknonip, 2019-11-28:
To clarify, my testing is all with docker-compose.

closed 2020-02-06 by brocaar