chirpstack/chirpstack-gateway-bridge #153: chirpstack-gateway-bridge debian package does not start on Raspberry PI 2

Niondir, 2020-01-08:
### Is this a bug or a feature request?
Probably a bug

### What did you expect?
Executing `/usr/bin/chirpstack-gateway-bridge` should start the bridge

### What happened?
```
pi@LoRaGateway:~ $ /usr/bin/chirpstack-gateway-bridge
Illegal instruction
```

### What version are your using?
```
pi@LoRaGateway:~ $ cat /etc/apt/sources.list.d/chirpstack.list
deb https://artifacts.chirpstack.io/packages/3.x/deb stable main
```

```
pi@LoRaGateway:~ $ apt-cache policy chirpstack-gateway-bridge
chirpstack-gateway-bridge:
  Installed: 3.5.0
  Candidate: 3.5.0
  Version table:
 *** 3.5.0 0
        500 https://artifacts.chirpstack.io/packages/3.x/deb/ stable/main armhf Packages
        100 /var/lib/dpkg/status
     3.4.1 0
        500 https://artifacts.chirpstack.io/packages/3.x/deb/ stable/main armhf Packages
     3.4.0 0
        500 https://artifacts.chirpstack.io/packages/3.x/deb/ stable/main armhf Packages
```

```
pi@LoRaGateway:~ $ uname -a
Linux LoRaGateway 4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 armv6l GNU/Linux
```

```
pi@LoRaGateway:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 900032
Serial          : 00000000eaf1ec6e
```

### How can your issue be reproduced?
Following this guide: https://www.chirpstack.io/guides/debian-ubuntu/

### Could you share your log output?
```
pi@LoRaGateway:~ $ sudo systemctl start chirpstack-gateway-bridge
pi@LoRaGateway:~ $ sudo journalctl -u chirpstack-gateway-bridge
-- Logs begin at Wed 2020-01-08 08:59:36 UTC, end at Wed 2020-01-08 08:59:43 UTC. --
Jan 08 08:59:40 LoRaGateway systemd[1]: Starting ChirpStack Gateway Bridge...
Jan 08 08:59:40 LoRaGateway systemd[1]: Started ChirpStack Gateway Bridge.
Jan 08 08:59:40 LoRaGateway systemd[1]: chirpstack-gateway-bridge.service: main process exited, code=killed, status=4/ILL
Jan 08 08:59:40 LoRaGateway systemd[1]: Unit chirpstack-gateway-bridge.service entered failed state.
```

```
pi@LoRaGateway:~ $ sudo systemctl status chirpstack-gateway-bridge
● chirpstack-gateway-bridge.service - ChirpStack Gateway Bridge
   Loaded: loaded (/lib/systemd/system/chirpstack-gateway-bridge.service; enabled)
   Active: failed (Result: start-limit) since Wed 2020-01-08 08:58:19 UTC; 1s ago
     Docs: https://www.chirpstack.io/
  Process: 18740 ExecStart=/usr/bin/chirpstack-gateway-bridge (code=killed, signal=ILL)
 Main PID: 18740 (code=killed, signal=ILL)
```

Niondir, 2020-01-08:
Same with:

````
wget https://artifacts.chirpstack.io/downloads/chirpstack-gateway-bridge/chirpstack-gateway-bridge_3.5.0_linux_armv7.tar.gz
tar -xzf chirpstack-gateway-bridge_3.5.0_linux_armv7.tar.gz

pi@LoRaGateway:~ $ ./chirpstack-gateway-bridge
Illegal instruction
```

Niondir, 2020-01-08:
Works with the armv6 package.

Can I somehow get the armv6 package from the debian repo? Or could it be added?

brocaar, 2020-06-12:
You must download the `_armv5.deb` files for Raspberry Pi 2.

closed 2020-06-12 by brocaar