chirpstack/chirpstack-gateway-bridge #156: Bridge stop working with Basic Station

istina-iot, 2020-01-17:
### Is this a bug or a feature request?
Bug
### What did you expect?
Normal execution without errors
### What happened?
After some time, bridge stop working with basic station without interrupting (our script can't reload bridge, because bridge still running)
### What version are your using?
`3.6`
### How can your issue be reproduced?
Run bridge with basic station several hours.
### Could you share your log output?
```
goroutine 24093 [chan receive, 3 minutes]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x350f2c0, 0x353c000, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c

goroutine 24096 [chan receive, 3 minutes]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x350f950, 0x353c0c0, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c

goroutine 24099 [chan receive, 3 minutes]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x350fa40, 0x353c240, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c

goroutine 24102 [chan receive, 3 minutes]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x350ff50, 0x353c300, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c

goroutine 24111 [chan receive, 3 minutes]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x3562600, 0x353c6c0, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c

goroutine 24108 [chan receive, 3 minutes]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x3562510, 0x353c540, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c

goroutine 24126 [chan receive, 3 minutes]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x35631d0, 0x353cc00, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c

goroutine 24114 [chan receive, 3 minutes]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x3562990, 0x353c780, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c

goroutine 24117 [chan receive, 3 minutes]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x3562ae0, 0x353c900, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c

goroutine 24120 [chan receive, 3 minutes]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x3562e10, 0x353c9c0, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
[... 37 lines cut ...]

goroutine 24141 [chan receive]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x350f590, 0x353c3c0, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c

goroutine 24144 [chan receive]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x350fd40, 0x353c600, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c

goroutine 24150 [chan receive]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x348e1e0, 0x353ccc0, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:746 +0x30
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c

goroutine 24153 [runnable]:
github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap.func2(0x348ef90, 0x353cf00, 0xc684d0)
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743
created by github.com/brocaar/chirpstack-gateway-bridge/internal/backend/basicstation.(*Backend).websocketWrap
        /chirpstack-gateway-bridge/internal/backend/basicstation/backend.go:743 +0x23c
```

brocaar, 2020-01-17:
I'll try to reproduce your issue. Thanks for reporting!

brocaar, 2020-01-23:
I believe the above commit should fix these errors. The issue was that the for loop would never return on an error.

commit faac69c7 (2020-01-23): In case of error, return from the Go routine.

Closes #156.
closed 2020-01-23 by brocaar