chirpstack/chirpstack-gateway-bridge #158: Support for Ursalink Gateway - GCP

patboul, 2020-01-21:
### Is this a bug or a feature request?
Feature request
### What did you expect?
Being able to install and configure the gateway bridge on an Ursalink gateway, with support for Google GCP as the back end. 
### What happened?

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2020-01-21:
Could you provide a link to the gateway you would like to see integrated? Do you work for Ursalink / would you be able to provide a gateway for testing? If so, could you contact me by e-mail to discuss the details (info@brocaar.com)?

patboul, 2020-01-21:
I am not working for Ursalink, we have 3 of their gateway we would like to
integrate in GCP.   With Ursalink support I got root access and notice that
Chirpstack is already installed. Once I got it working and with their
permissions I will share my findings.

Hold it for now.

Thanks.

brocaar, 2020-02-06:
Ok, that is great :-) Would you mind creating a pull-request to add these setup steps to the documentation for the Ursalink?

patboul, 2020-02-06:
I brocaar,

I suspended work on this because of a bug I encountered with the network-server part ( a bad setup in GCP or chirsptack-gateway, I don't know ). My setup was woking with one gateway ( Dragino ) and data was flowing. I could see activity in Chirptack App and on GCP. The gateway object in google was showing heatbeat activity and telemetry event.

When I added a second gateway, another Dragino or the Ursalink, in GCP, I could see heartbeat but no telemetry. In Chirpsack app, the gateway never showed any activity.

As we are still in the proof of concept phase and not on a tight schedule,  I temporarily work around this problem by using my own MQTT broker (mosquitto) inside the VM.  I configured the network server backend to use this. I configured the gateways to connect using mqtt over tls on my vm. It should be easy to switch the backend when a solution is found or still, use mosquitto as it doesn't seem to consume much ressources and security is enforce with TLS. 

As my lab environment was changed a lot and is potentially in a bad state, I will rebuild it and retry when I get time. I will of course share my finding.

In the mean time, I suggest  to put this issue on hold or close it. I will come back when I get more information.

closed 2020-02-13 by brocaar