chirpstack/chirpstack-gateway-bridge #159: TerminateOnConnect not working with MqTT Autoreconnect

schmo90, 2020-02-03:
### Is this a bug or a feature request?

Mqtt is not correctly reconnecting again after lost of internet connection

### What did you expect?

Application is closing

### What happened?

Application is running in a endless loop

### What version are your using?

 3.6

### How can your issue be reproduced?

tc qdisc add dev 3g-GPRS root netem loss 100% 100%

### Modified Source Code

in backend.go
func NewBackend(conf config.Config) (*Backend, error) {
.
.
.
	b.clientOpts.SetAutoReconnect(false) 
.
.

backend.go

func (b *Backend) onConnectionLost(c paho.Client, err error) {
	mqttDisconnectCounter().Inc()
	log.WithError(err).Error("mqtt: connection error")
    if b.terminateOnConnectError {
				log.Fatal(err)
	}

	log.WithError(err).Error("integration/mqtt: connection error")
	time.Sleep(time.Second * 2)
}

brocaar, 2020-02-06:
Thanks. Instead of providing source modifications in an issue, would you mind creating a pull-request?

paolofrs, 2020-08-29:
Hi, has this bug been fixed? I have a Dragino LPS8 Gateway at a rural location with bad internet coverage and I have to manually restart the gateway every time internet connection gets lost because the app do not reconnect to the MQTT broker. 

I am using version 3.8.0 of gateway server .ipk [download here](https://artifacts.chirpstack.io/vendor/dragino/LG308/). App and broker use MQTT over TLS (I don't know if this can make any difference).

paolofrs, 2020-09-02:
I've found [this discussion](https://forum.chirpstack.io/t/gateway-local-persistence-for-internet-outages/1747) and [this reply[(https://forum.chirpstack.io/t/gateway-local-persistence-for-internet-outages/1747/12). 

[This fix](https://github.com/brocaar/chirpstack-gateway-bridge/commit/c5536c663f9d587e97e2c908c33ad87534b48976) is supposed to fix the reconnection problem as well avoiding the endless loop?

gbird3, 2021-01-14:
From my testing, the terminateOnConnect seems to only work when the gateway-bridge is starting up. If the mosquitto connection dies while the gateway-bridge is running, it is not terminating it. Is this the way it is designed?

For my use case, it would be really helpful to have it terminate after startup. I have been running into issues where it seems like the mqtt connection is dying and the gateway-bridge can't reconnect. In this case, as soon as I restart the gateway-bridge everything starts working again.

brocaar, 2021-01-28:
I have just merged a fix for this.

closed 2021-01-28 by brocaar