chirpstack/chirpstack-gateway-bridge #168: New kerlink toolchain and architecture

Regimbal, 2020-04-05:
### Is this a bug or a feature request?
Feature request or improvement

### What did you expect?
New Kerlink product range and related firmware rely on a new toolchain/architecture named klkgw

### What happened?
I have a new Kerlink iStation and no chirpstack-gateway-bridge package is available

### What version are your using?
Starting KerOS 4.1, all packages shall be compiled with the new toolchain provided and architecture named klkgw.
Havent seen any corresponding package since 3.7.1 version of Chirpstack-bridge-gateway

### What I did
I tried compiling Chirpstack-bridge-gateway and package it with the kerlink toolchain.
That seems to work correctly

I can provide what i did in a pull request if needed. But I did nothing more than adding the "istation" folder under kerlink, and corresponding files.

brocaar, 2020-04-09:
@Regimbal could you give this `.ipk` a try? https://artifacts.chirpstack.io/vendor/kerlink/istation/

I have also updated the documentation in https://github.com/brocaar/chirpstack-gateway-bridge/commit/f8e745726f485b6b623f5b5c80eed77911a01ece.

Regimbal, 2020-04-14:
Dear Brocaar,
It seems to work correctly, yet I observed an issue which is probably not related to this directly
In the config file : 
- if UDP is binded to 0.0.0.0:1700 it works correctly
- If binded to localhost:1700 or 127.0.0.1:1700, nothing is sent to MQTT.
In both cases "tcpdump -AUq -i lo port 1700" returns correct events

brocaar, 2020-04-14:
@Regimbal that seems to be unrelated and I do not have an answer to that question as I don't have a iStation to test with.

commit 53adcb40 (2020-04-15): Add scripts for Kerlink iStation packaging.

Closes #168.
closed 2020-04-15 by brocaar