chirpstack/chirpstack-gateway-bridge #174: Dragino LPS8 gateway support

aalbinati, 2020-08-03:
- [X] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## Summary

As the Dragino LPS8 gateway is becoming increasingly popular over the last months. It would be nice to have a package built for it. It shouldn't be so different from the LG308 package since both use the same architecture (AR9331 24K MIPS / Linux). The main difference being that the LG308 uses a SX1301 chip and the LPS8 uses a SX1308 chip.

## Can you implement this by yourself and make a pull request?

I don't know how to tweak the LG308 package to make it LPS8 compatible but I can actively help in testing the functionality since I already have several devices and a running Chirpstack Server stack.

[Dragino gateways comparison](https://www.dragino.com/downloads/downloads/LoRa_Gateway/Dragino%20LoRa%20Gateway%20Compare%20List.pdf)

[LPS8 Datasheet](https://www.dragino.com/downloads/index.php?dir=LoRa_Gateway/LPS8/&file=Datasheet_LPS8_LoRaWAN%20Pico%20Station.pdf)

[LPS8 source code](https://github.com/dragino/openwrt_lede-18.06)

Thanks!

paolofrs, 2020-08-29:
Hi, I use LPS8 with ChirpStack Gateway installed on it (used this procedure for [LG308).](https://www.chirpstack.io/gateway-bridge/gateway/dragino/). So far everything is ok! I use MQTT over TLS to forward packets from gateway to network server.

As you said, the only difference is in the (let's call it so) Network Interface Controller. I believe that both the NICs' drivers exposes the same functionalities to the OS and the package is built for the same release of OWRT (since target is the same). Thus, from the point of view of the gateway app there is no hardware difference and it works on both devices. 

What do you think?

aalbinati, 2020-09-08:
Hi @paolofrs, I am trying to get the gatway bridge to run in the LPS8 but the TCP packets never reach the server. Have you managed to get the process to log its information? I don't know how to see it.

paolofrs, 2020-09-23:
Hi @aalbinati, I am sorry it does not work for you. Let's see... have you followed all the installation steps correctly (i.e. download in separate folder like /tmp then remove opkg)? Did you set up the packet forwarder correctly (via the dragino gui available @ http://lps8_ip:8000)? Which version of   Dragino OS are you using? If u are on the last you should see the gw app's log output on web console log (under the system tab i guess). Just enable "all" logs. If you connect via ssh you can use [logread](https://openwrt.org/docs/guide-user/base-system/log.essentials)

paolofrs, 2020-09-23:
I assume you have already tested the connection to the server from the lps8's lan, no "un-ruled" firewalls in between and all ports opened (1883 for mqtt and 8883 for mqtts)...

aalbinati, 2020-09-23:
Hi @paolofrs , I don't seem to find any setting where I can change the logging level to "all". Even logread via ssh doesn't seem to log the chirpstack gateway app. I am using version lgw-5.4.1599149041 which was released a few days ago.

I will check if the dargino firewall has the ports closed for MQTT but that still doesn't explain why I am not seeing any logs.

The service seems to start and stop correctly.

Any guess why these could be happening or where I can enable the logs for the service? Thanks!

paolofrs, 2020-09-23:
The thing I am stuck  with is this [issue](https://github.com/brocaar/chirpstack-gateway-bridge/issues/159). But when the connection is stable the gw app on lps8 run smoothly (more or less... i ve noticed some messages to get not ack by the gw to the server while they were sent to end node instead)

s1116162, 2020-10-03:
> Hi @paolofrs , I don't seem to find any setting where I can change the logging level to "all". Even logread via ssh doesn't seem to log the chirpstack gateway app. I am using version [...]

I am having the exact same issue on the LPS8 using the latest firmware and gateway bridge build. Nothing gets logged at all. I don't even get output when running the gateway bridge manually.

aalbinati, 2020-10-04:
Since it seems I am not the only one experiencing this problem. @brocaar could you give it a look please? I am happy to collaborate in the solution if you need my help. I can also provide more information about the issue if you need it.

Thanks!

brocaar, 2020-10-05:
@aalbinati I do not have a LPS8 gateway to test with currently.

paolofrs, 2020-12-22:
> Hi @paolofrs , I don't seem to find any setting where I can change the logging level to "all". Even logread via ssh doesn't seem to log the chirpstack gateway app. I am using version [...]

Hi, @aalbinati sorry for the very late answer... I had time to make some more test and indeed... I discovered that latest firmware versions (from lgw-5.4.1599149041 ongoing) are not any more compatible with chirpstack-gateway-bridge released for LG308. I do not know why. 

The last firmware version that worked to me was `LG02_LG08--build-v5.4.1593407804-20200629-1318` (they should have made some big changes since the filename changes as well from that version). I've just sampled in a lucky spot... 

I can tell that that fw version works with `chirpstack-gateway-bridge_3.9.0-r1_mips_24kc.ipk` and have no informations for newers (I will try to revert to old working LPS8 firmware and latest gateway package).

@brocaar if this can help: when running chirpstack gateway either with `/etc/init.d/chirpstack-gateway-bridge start` or directly with `/opt/chirpstack-gateway-bridge/chirpstack-gateway-bridge --config <path/to/chirpstack-gateway-bridge.toml>` the process gets a PID and I can see it with `top`. It uses a huge amount of CPU (compared to dragino pkt forwarder and other process). At first I assumed a resource issue but it does not make that much sense... It seems that the packets are not corretly forwarded to the gateway. There is any way to check if is either the gateway not working at all or the packet forwarder not properly routing UDP packets to the proper socket where cs-gateway is listening?

aalbinati, 2020-12-24:
@paolofrs I see. 
The changes made in the firmware that broke the chirpstack-gateway-bridge are the following:

Link: [LPS8 Changelog](https://www.dragino.com/downloads/downloads/LoRa_Gateway/LPS8/Firmware/ChangeLog)
lgw--build-v5.4.1598286158-20200825-0025
  --Rename firmware form LG02-LG08 to  lgw
  --Remove Luci packets
  --LG01/LG02 Pkt_Fwd to use new UCI structure
  --Fix LG02 pkt_fwd hung bug
  --DLOS8 GPS position extend to 6 digits. 
  --Better view for LoRaWAN page
  --Remove HTTP for LG308/LPS8/DLOS8
  --Change Server address for TTN AU915 / AS923
  --Enable Manual Config GPS (fake latitude, longitude and altitude) 
  --System Overview page now shows if there is Cellular module
  --Add Kernal Patch in the build environment to build
  --Add Wifi for stable boot

Maybe you can identify a possible issue with one of those changes
For the time being, I'll be forwarding the packets directly to the server.

@brocaar If you want us to something to fix the issue just tell us. We are here to help!
Happy Holidays!

paolofrs, 2020-12-27:
Hi @aalbinati ,

I've found out that the latest working release of chirpstack-gateway-bridge package for LPS8 is [chirpstack-gateway-bridge_3.8.0-r1_mips_24kc.ipk](https://artifacts.chirpstack.io/vendor/dragino/LG308/chirpstack-gateway-bridge_3.8.0-r1_mips_24kc.ipk). 

This package works with the latest release of Dragino OpenWRT distro for LPS8/LG02, [lgw--build-v5.4.1607519907-20201209-2120](https://www.dragino.com/downloads/downloads/LoRa_Gateway/LPS8/Firmware/Release/lgw--build-v5.4.1607519907-20201209-2120/).

I will try to set up a cross-compile environment to build from chirpstack latest sources a package that is compatible with LPS8. I am quite new to these stuff, and it doesn't look that easy... Any help is more than welcome!

Happy Holidays! Buone Feste! :)

Haridhra, 2020-12-29:
Hi all,

I am in the process of deciding between Dragino LPS8 or a Dragino LG308 LoRaWAN Gateway. Only thing stopping me from pulling the trigger on the much cheaper LPS8 is questionable compatibility with the chirpstack-gateway-bridge running on the gateway.

@paolofrs  Does the version of the Chirpstack gateway bridge that you mention (3.8.0-r1) work with the latest version of Dragino OpenWRT distro when I follow the steps [here](https://www.chirpstack.io/gateway-bridge/gateway/dragino/) ? Or did you have to do something different to get it working? Is it fully functional or only partially functional with limited features working? It would be of great help if you could share any information you know about this.

In case it is not possible to get chirpstack gateway bridge running on the LPS8 gateway, would it be possible to get the chirpstack gateway bridge running on a cloud server and use UDP packet forwarder on the Dragino LPS8 to get the LoRaWAN packets to ChirpStack?  If this is the case, then please point me in the right direction to configure the UDP packet forwarder to work with chirpstack gateway bridge running on a cloud server.

Any help would be very useful to me!

Happy holidays!

Thanks in advance,
Harid.

aalbinati, 2020-12-29:
Hi @Haridhra,

It is possible to forward the UDP packets to a server instance of the chirpstack gateway bridge. It's actually what I am doing nowadays until we get a fix. It is explained in this link: https://forum.chirpstack.io/t/dragino-gateway/9082/7

aalbinati, 2020-12-29:
@paolofrs did you make any progress in the cross-compilation? It would be great if you could explain me why do you think the version compiled by you will be different from the one available for download.

Haridhra, 2020-12-29:
Hi @aalbinati ,

Thanks for the quick reply! That's good to know! Hopefully we get a fix soon! Running the Gateway Bridge on the gateway makes the solution much more secure after all.

Do you know any downsides to running the older version of the gateway bridge that @paolofrs mentioned on the gateway? Would that be a better way to go until we have the latest version working compared to UDP?

aalbinati, 2020-12-29:
I don't know for sure what changes were made since that version so I can't tell. UDP forwarding to the server is not unsecure, packets are still encrypted by LORAWAN protocol that makes sure all information is secure. TCP forwarding is better since it makes sure that the packets get to the server. The gateway-bridge also adds one more layer of security by providing an ssl connection between the bridge and the server but the packets will be encrypted anyway so it wouldn't be a vulnerability to forward them directly to the server.

As far as I know we don't have any reason to believe UDP forwarding would be a bad solution.

Haridhra, 2020-12-29:
> I don't know for sure what changes were made since that version so I can't tell. UDP forwarding to the server is not unsecure, packets are still encrypted by LORAWAN protocol that makes sure all information is secure. TCP forwarding is better since it makes sure that the packets get to the server. The gateway-bridge also adds one more layer of security by providing an ssl connection between the bridge and the server but the packets will be encrypted anyway so it wouldn't be a vulnerability to forward them directly to the server.
> 
> As far as I know we don't have any reason to believe UDP forwarding would be a bad workaround until the issue is fixed.

Ah I see, thanks a lot for the information! I think I can comfortably use UDP until we get a fix then!

aalbinati, 2020-12-29:
> Hi @aalbinati ,
>
> I've found out that the latest working release of chirpstack-gateway-bridge package for LPS8 is [...]

Hi @Haridhra @paolofrs ,

I can confirm I had the same results. Version 3.8.0 works fine in Dragino LPS8 with its latest firmware!
If we want to keep compatibility with future version we should try compiling the code by ourselves and change whichever part of the code messes up compatibility with LPS8. I'm sure its just a tiny fraction of code that breaks it, the difficult part is to find it if we didn't code it.

paolofrs, 2020-12-30:
As far as I know, the UDP isn't secure but even worse it isn't reliable... Running CS Gateway Bridge on the LPS8/LG308 enable MQTT(S) protocol that is secure and reliable (if it works properly 😄 ) . It uses TCP with SSL encryption and authentication. What @aalbinati says about the lora packets it's true but... you share all the rest when using UDP. That would make a bored guy who want to sniff the temperature of your country house very happy... 😄 

@Haridhra I just want to add this thing I've recently discovered: I suggest you to use the LPS8 ETH port to access internet. Using the WiFi Client mode is not very stable at the moment. I assume an RF hardware problem but we will see... I am really happy of the LPS8 and Dragino products in general (I am using LT I/O controller) thus both of them should be ok! 

@aalbinati good to heard that!  I've contacted Dragino directly to see how to compile GO for LPS8. 👍

aalbinati, 2020-12-30:
Hi @paolofrs,

I can manage to get the bridge working on the LPS8 with bridge version 3.8.0 but have some strange error that blocks the downlink packets.

Tue Dec 29 22:49:31 2020 user.info chirpstack-gateway-bridge[10736]: time="2020-12-30T01:49:31Z" level=info msg="integration/mqtt: downlink frame received" downlink_id=da946afd-e4d2-4c8e-8dc8-a4e1db22441d gateway_id=0000000000000000

Tue Dec 29 22:49:31 2020 user.err chirpstack-gateway-bridge[10736]: time="2020-12-30T01:49:31Z" level=error msg="send downlink frame error" error="get gateway error: gateway does not exist"

My gateway id isn't 0000000000000 so something must be wrong there.
If I just forward the UDP packets to the server bridge it works fine, so it is a really strange error don't you think?

Any help would be appreciated

aalbinati, 2020-12-30:
When inspecting the downlink packet i see this.

gateway/a840411dab044150/command/down {"phyPayload":null,"txInfo":null,"token":39753,"downlinkID":"pU935k1KQPOFkt6S+k15NQ==","items":[{"phyPayload":"IHrMhC7uCHBSIRFJmvJwM3Frbzxb4DnhFYCAE5sBT9Wy","txInfo":{"gatewayID":null,"frequency":923300000,"power":27,"modulation":"LORA","loRaModulationInfo":{"bandwidth":500,"spreadingFactor":12,"codeRate":"4/5","polarizationInversion":true},"board":0,"antenna":0,"timing":"DELAY","delayTimingInfo":{"delay":"5s"},"context":"nnb9FA=="}},{"phyPayload":"IHrMhC7uCHBSIRFJmvJwM3Frbzxb4DnhFYCAE5sBT9Wy","txInfo":{"gatewayID":null,"frequency":923300000,"power":27,"modulation":"LORA","loRaModulationInfo":{"bandwidth":500,"spreadingFactor":12,"codeRate":"4/5","polarizationInversion":true},"board":0,"antenna":0,"timing":"DELAY","delayTimingInfo":{"delay":"6s"},"context":"nnb9FA=="}}],"gatewayID":"MygwID base64 encoded"}

There's something odd with the downlink message having gatewayID:null don't you think?

aalbinati, 2020-12-30:
When inspecting the downlink packet i see this.

gateway/*myGWid*/command/down {"phyPayload":null,"txInfo":null,"token":39753,"downlinkID":"pU935k1KQPOFkt6S+k15NQ==","items":[{"phyPayload":"IHrMhC7uCHBSIRFJmvJwM3Frbzxb4DnhFYCAE5sBT9Wy","txInfo":{"gatewayID":null,"frequency":923300000,"power":27,"modulation":"LORA","loRaModulationInfo":{"bandwidth":500,"spreadingFactor":12,"codeRate":"4/5","polarizationInversion":true},"board":0,"antenna":0,"timing":"DELAY","delayTimingInfo":{"delay":"5s"},"context":"nnb9FA=="}},{"phyPayload":"IHrMhC7uCHBSIRFJmvJwM3Frbzxb4DnhFYCAE5sBT9Wy","txInfo":{"gatewayID":null,"frequency":923300000,"power":27,"modulation":"LORA","loRaModulationInfo":{"bandwidth":500,"spreadingFactor":12,"codeRate":"4/5","polarizationInversion":true},"board":0,"antenna":0,"timing":"DELAY","delayTimingInfo":{"delay":"6s"},"context":"nnb9FA=="}}],"gatewayID":"*MygwID base64 encoded*"}

There's something odd with the downlink message having gatewayID:null don't you think?

Haridhra, 2020-12-31:
> As far as I know, the UDP isn't secure but even worse it isn't reliable... Running CS Gateway Bridge on the LPS8/LG308 enable MQTT(S) protocol that is secure and reliable (if it works properly 😄 ) . [...]

Thanks @paolofrs  for the information! So as I understand it, the LoRaWAN encryption keeps the data from the node to the gateway safe but UDP could compromise gateway to LoRaWAN server (cloud) security. Do I have it right?

If yes, I was reading about the LoRa Basics Station and that seems to be safer than UDP. Is there a way we can get Basics Station running on the LPS8?

pulidoj, 2021-01-02:
Hi all of you,

It's really interesting your discussion about this issue.
I'm also working on this as we use a lot of Dragino products: LG308, LPS8, OLG01, DLOS8...
So far we have been working with UDP mode over Gateway Bridge installed on servers and everything works really great.

We are starting to migrate to Gateway Bridge installed on gateways because of security and also for some blockchain projects and our results are the same as yours for the moment, for both LG308 and LPS8.
Version 3.8.0 works great, but neither 3.9.1 nor 3.9.2 do it.
For some reason the lora_pkt_fwd cannot connect to Gateway Bridge in those versions.

We are working on this and we will let you know if we get some positive results.
If you get something and need to test it in any Dragino products, just let us know.

Related to @Haridhra request about Basic Station, I guess it could be a parallel product to Chirpstack Gateway Bridge. I had no much time to research, but perhaps Chirpstack Network server should need some kind of adaptation to it. I'll have a look more in depth.

Regards,

Jesus

Haridhra, 2021-01-02:
@pulidoj  I see! Thanks for the information!

Meanwhile while I try to get more information about Basic Station, I would like to know if UDP is okay to use for commercial applications or if it's not at all recommended for commercial applications?

I see that other network servers like The Things Stack don't have an offering for gateway bridges running on gateways. If UDP is insecure would they not try to steer away from it? Or does running the gateway bridge on the gateway just provide an additional layer of security? I'm quite new to network security, so please bear with my questions. :)

pulidoj, 2021-01-02:
Hi @Haridhra,

Well, I think it depends very much on the scope of the project, the type of the network.
Our experience is that we have more than 250 gateways running in UDP mode and you had no security issues for the moment, the first ones running with TTN as well.

But I think it is not just say UDP is secure or insecure. I would consider different aspects:
- If LoRa devices work in OTAA or ABP mode. Depends on the project it is more interesting one out of the two modes. OTAA mode adds a security layer as creates keys when makes the Join process but in some cases this Joining process can produce some drawbacks.
- If you use public or private LoRa networks. Private networks need more management but you can implement custom firewall systems, and also small private networks are more difficult to find and be worth to be hacked.
- The number of devices are increasing very very quickly and the risk of bugs in some of them (I can sure you we found lots of them) will make security more important every day.
- UDP don't allow the possibility of Certificates, a very robust system to restrict the devices and gateways that can connect to a LoRa server.
- The architecture of the project is also important. It is not the same LoRa networks where gateways work only as packet forwarder pointing to a cloud LoRa Server, where the application server is on the cloud, that a LoRa server installed on a RPi or even some gateways with Nodered embedded where all the logic of application can be developed inside the LAN network, and they even don't need to go up to internet.

So, as conclusion, I think gateway bridge is a great tool to have in hand and use it when it suits better.

Regards.

ionultd, 2021-03-01:
Hi @pulidoj , @Haridhra , @paolofrs , @aalbinati any updates on getting the Dragino LPS8 to work with Chirpstack? My firmware version is lgw-5.4.1610176959 and I do have the gateway ipk running on the gateway but for some reason when I set the Lorawan settings via the Dragino admin it appears to not accept them. I put the correct settings in and when submitted the form comes back  empty as if I had no settings set. Image 1 of the settings in then image 2 after submitting the settings. 
[Image 1](https://i.imgur.com/WmRWyMe.png)
[Image 2](https://i.imgur.com/j8zbs5j.png)

aalbinati, 2021-03-03:
> Hi @pulidoj , @Haridhra , @paolofrs , @aalbinati any updates on getting the Dragino LPS8 to work with Chirpstack? My firmware version is lgw-5.4.1610176959 and I do have the gateway ipk running on [...]

I don't know why that's happening to you. I am using the LPS8 gateway with no problems but I couldn't get it to work with the gateway bridge installed in it.
Your problem doesn't have nothing to do with that. Try, using another bowser or switch to version lgw-5.4.1607519907 which works fine for me.

aalbinati, 2021-03-17:
> I have been able to make Gateway Bridge work on the LPS8.
> 
> The steps that I used are the following:
> Updated the Gateway firmware to the latest version “dragino-lgw--v5.4.1610176959-squashfs-sysupgrade.bin”.
> Install the latest Gateway Bridge “chirpstack-gateway-bridge_3.10.0-r1_mips_24kc.ipk”.
> At this point the Gateway Bridge will not work.
> Get the latest code for the Gateway Bridge from the Git repository.
> Do a “build snapshot” to compile it for the non-native supported platforms.
> I used an Ubuntu VM and needed to install some dependencies.
> Get the file “chirpstack-gateway-bridge” from “dist/linux_mips” and replace the one in the Gateway with it (“/opt/chirpstack-gateway-bridge/”).
> Now it will work.
> 
> Maybe an updated ipk is needed?

That's great! Could you share the compiled package please?

pduarte-coder, 2021-03-18:
> That's great! Could you share the compiled package please?

The build process does not produce any ipk file, just deb, rpm, tar.gz and the executable (that I have attached).
For the record, at the time I build it, it was chirpstack-gateway-bridge_SNAPSHOT-99d1450_linux_mips.

[chirpstack-gateway-bridge.zip](https://github.com/brocaar/chirpstack-gateway-bridge/files/6162913/chirpstack-gateway-bridge.zip)

aalbinati, 2021-03-18:
Perfect, I will try to install it later today!

aalbinati, 2021-03-19:
Tried updating to the latest firmware but now my gateway is bricked :(
I just sent an email to Dragino but if anyone knows how to unbrick it I would really appreciate you tell me.
It only has the red power led on. It doesn't broadcast any wifi and the Ethernet connection fails.

aalbinati, 2021-03-19:
> Tried updating to the latest firmware but now my gateway is bricked :(
> I just sent an email to Dragino but if anyone knows how to unbrick it I would really appreciate you tell me.
> It only has the red power led on. It doesn't broadcast any wifi and the Ethernet connection fails.

Solved!!! If anyone encounters the same problem head to [this guide](https://wiki.dragino.com/index.php?title=Recover_Gateway). It really helped me out!

aalbinati, 2021-03-19:
> The build process does not produce any ipk file, just deb, rpm, tar.gz and the executable (that I have attached).
> For the record, at the time I build it, it was [...]

I did everything exactly as you mentioned but with the newest firmware which is dragino-lgw--v5.4.1614945073-squashfs-sysupgrade.bin. Unfortunately it didn't work. Don't know what might be happening.

pduarte-coder, 2021-03-19:
I have used this procedure with 3 Gateways, and they are working correctly.
Did you unzip the file, and after copying it to the gateway, changed its permissions?
Did you changed the “Primary LoRaWAN  Server” and edited de Gateway Bridge configuration file?
What steps have you executed to check that “it didn’t work”?

aalbinati, 2021-03-19:
> I have used this procedure with 3 Gateways, and they are working correctly.
> Did you unzip the file, and after copying it to the gateway, changed its permissions?
> Did you changed the “Primary LoRaWAN [...]

I have done all of that except for the file permissions part. I will try with permissions set on 777. Thanks

aalbinati, 2021-03-19:
I tried that but it doesn't even execute. Whenever I run `/etc/init.d/chirpstack-gateway-bridge start` it says _Starting chirpstack-gateway-bridge_ meaning it is running the init file but the process really never starts because I cant't find it using ps command and when I type `/etc/init.d/chirpstack-gateway-bridge stop` it says:

 _Stopping chirpstack-gateway-bridge
start-stop-daemon: warning: killing process 27026: No such process_

Weird don't you think?
Do you know what could be causing this issue?

bmurithi, 2021-07-19:
For anyone looking for a device-specific solution to this, thje OpenWRT packages feed ships with golang package support.
(On an 18.04) based build, I had to pull in golang package support from the mainline. Used existing golang WRT packages in the mainline feeds to set this up.

Had to compile a working version with the toolchain to the the bridge to work on a dragino with the unified firmware (lgw).
You can modify this to include a configuration package and procd configuration for supervision.

## Building

```
# inside a working buildroot (at the very least initialized/prepared)
cd openwrt
cd feeds/packages
git apply packages.diff
cd ../../
./scripts/feeds update -a
./scripts/feeds install chirpstack-gateway-bridge
make menuconfig
# Select chirpstack-gateway-bridge under Network > LoRa
# Save your changes to build configuration
 make package/chirpstack-gateway-bridge/compile
```

You can customize the package manifest to include a default configuration and procd setup so it's supervised.

```
vagrant@ubuntu-bionic:~/dragino/dragino-lede-18.06/openwrt$ cd bin/packages/mips_24kc/
base/      dragino/   luci/      luci2/     packages/  routing/   telephony/ xluci2/    
vagrant@ubuntu-bionic:~/dragino/dragino-lede-18.06/openwrt$ find bin/packages/mips_24kc/packages/chirp*
bin/packages/mips_24kc/packages/chirpstack-gateway-bridge_3.11.0-1_mips_24kc.ipk
vagrant@ubuntu-bionic:~/dragino/dragino-lede-18.06/openwrt$ 

```

## Patch + Package

:card_file_box:  [chirpstack-gateway-bridge-dragino-build-patch.zip](https://github.com/brocaar/chirpstack-gateway-bridge/files/6841923/chirpstack-gateway-bridge-dragino-build-patch.zip)

## Running on a dragino pico gateway (LPS8)

[image]

bmurithi, 2021-07-19:
> I tried that but it doesn't even execute. Whenever I run `/etc/init.d/chirpstack-gateway-bridge start` it says _Starting chirpstack-gateway-bridge_ meaning it is running the init file but the [...]

Replace `/etc/init.d/chirpstack-gateway-bridge` with a procd-based configuration (sample -below)

> My `PROG` path is different as I compiled the gateway bridge using the OpenWRT golang toolchain.

```
#!/bin/sh /etc/rc.common

START=99

USE_PROCD=1
PROG=/opt/chirpstack-gateway-bridge/chirpstack-gateway-bridge
CONFIG=/opt/chirpstack-gateway-bridge/chirpstack-gateway-bridge.toml

start_service() {
    procd_open_instance
    procd_set_param command "$PROG"
    procd_append_param command --config "${CONFIG}"
    procd_set_param stderr 1
    procd_set_param stdout 1
    procd_set_param respawn
    procd_close_instance
}
```

# Starting / Stopping

```
root@dragino:/# ubus call service list '{"name": "chirpstack-gateway-bridge"}'
{
	"chirpstack-gateway-bridge": {
		"instances": {
			"instance1": {
				"running": true,
				"pid": 10582,
				"command": [
					"\/usr\/bin\/chirpstack-gateway-bridge",
					"--config",
					"\/opt\/chirpstack-gateway-bridge\/chirpstack-gateway-bridge.toml"
				],
				"term_timeout": 5,
				"respawn": {
					"threshold": 3600,
					"timeout": 5,
					"retry": 5
				}
			}
		}
	}
}
root@dragino:/# /etc/init.d/chirpstack-gateway-bridge stop
root@dragino:/# ubus call service list '{"name": "chirpstack-gateway-bridge"}'
{
	
}
root@dragino:/# pgrep chirpstack
root@dragino:/# /etc/init.d/chirpstack-gateway-bridge start
root@dragino:/# pgrep chirpstack
12120
root@dragino:/# ubus call service list '{"name": "chirpstack-gateway-bridge"}'
{
	"chirpstack-gateway-bridge": {
		"instances": {
			"instance1": {
				"running": true,
				"pid": 12120,
				"command": [
					"\/usr\/bin\/chirpstack-gateway-bridge",
					"--config",
					"\/opt\/chirpstack-gateway-bridge\/chirpstack-gateway-bridge.toml"
				],
				"term_timeout": 5,
				"respawn": {
					"threshold": 3600,
					"timeout": 5,
					"retry": 5
				}
			}
		}
	}
}
root@dragino:/# 

```

thienvomybk0507, 2021-09-13:
I need some document about : Setup Dragino Lps8 LoRaWAN Gateway and adding sensors to Chirpstack Network Server. Who can give me?

bmurithi, 2021-09-13:
> I need some document about : Setup Dragino Lps8 LoRaWAN Gateway and adding sensors to Chirpstack Network Server. Who can give me?

See: https://wiki.dragino.com/index.php?title=Notes_for_ChirpStack

Usually straightforward for most use-cases.

tmedicci, 2022-03-28:
Just an update: the newest version package for the [LG308](https://artifacts.chirpstack.io/vendor/dragino/LG308/chirpstack-gateway-bridge_3.13.2-r1_mips_24kc.ipk) still hangs on the LPS8 model.

I've built the v3.13.2 for the LPS8 myself: [chirpstack-gateway-bridge_3.13.2-1_mips_24kc.zip](https://github.com/brocaar/chirpstack-gateway-bridge/files/8364964/chirpstack-gateway-bridge_3.13.2-1_mips_24kc.zip)

brocaar, 2023-01-26:
Please refer to https://www.chirpstack.io/docs/chirpstack-mqtt-forwarder/install/dragino.html. Together with one of my clients, I've been working on a re-implementation of the ChirpStack Gateway Bridge, but then specific for the gateway with as requirement that it must run on devices with limited resourced (like the Dragino OpenWRT based gateways).

closed 2023-01-26 by brocaar