chirpstack/chirpstack-gateway-bridge #177: Filtering

dsMartyn, 2020-09-10:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## Summary

NetID and JoinID filtering individually 

The current code for filtering is on both NetID and JoinEUI.
`	if len(netIDs) == 0 && len(joinEUIs) == 0 {
		return true
	}`

Is it possible to have either instead of both?  We currently don't have our own netID assigned by LoRaAlliance so would like to leave the NetID open to all but limit the joins based on our Join EUI MAC. 

## What is the use-case?

To filter on NetID requires a subnet assigned from LoRa Alliance and it costs to get a large subnet.

This would allow filtering on either netID / JoinEUI.  Also to pave the way for allowing filtering based on DevEUI in the future. If the GW bridge could sync a list of DevEUI's from the network periodically or upon devEUI being added to the NWS then we could filter our own devices and hugely reduce the backhaul bandwidth.  
## Implementation description

probably

brocaar, 2020-09-10:
https://github.com/brocaar/chirpstack-gateway-bridge/blob/master/internal/filters/filters.go#L58

```go
	// return true when no filters are configured
	if len(netIDs) == 0 && len(joinEUIs) == 0 {
		return true
	}
```

I believe the part of code you are referring to, should work correctly. In case there are no NetID filters **and** there are no JoinEUI filters, then `MatchFilter` returns true, meaning the uplink will pass without filtering.

In any other case, the filters are applied (either NetID filters are set, JoinEUI filters are set or NetID and JoinEUI filters are set).

dsMartyn, 2020-09-10:
agh! sorry, i see it now my bad.

closed 2020-09-10 by dsMartyn