chirpstack/chirpstack-gateway-bridge #179: Add Basic Station Time Synchronization

vfylyk, 2020-09-21:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## Summary

It seems like the current Basic Station implementation does not include support for the [LNS @timesync@](https://doc.sm.tc/station/tcproto.html#time-synchronization) message type, so it would be great if it could be included.

This is what happens currently:

```
time="2020-09-21T04:21:24Z" level=debug msg="backend/basicstation: message received" gateway_id=d4adbdfffe9e2890 message="{\"msgtype\":\"timesync\",\"txtime\":4035744838}"
time="2020-09-21T04:21:24Z" level=info msg="backend/basicstation: raw packet-forwarder event received" gateway_id=d4adbdfffe9e2890 raw_id=d145516d-901f-412b-b65c-57bfa0a8d6bb
time="2020-09-21T04:21:24Z" level=info msg="integration/mqtt: publishing event" event=raw qos=1 raw_id=d145516d-901f-412b-b65c-57bfa0a8d6bb topic=gateway/d4adbdfffe9e2890/au/event/raw
```

## What is the use-case?

The most recent Cisco IXM gateway comes with Basic Station support, using GPS and @timesync@ messages to provide timestamps. Whenever possible I prefer to have timestamps applied at the gateway as they more accurately tell when a payload has been received.

## Implementation description

The [LNS Protocol](https://doc.sm.tc/station/tcproto.html#time-synchronization) indicates how this is implemented.

## Can you implement this by yourself and make a pull request?

I can give it a try.

brocaar, 2020-09-28:
If you would like to create a PR for this, that would be much appreciated :+1: 

Where the LNS Protocol docs mention that the LNS must answer, I think it makes sense to let the ChirpStack Gateway Bridge respond with the gps timestamp.

The `github.com/brocaar/lorawan/gps` package provides an utility to convert from a `time.Time` to GPS time (`time.Duration`): https://pkg.go.dev/github.com/brocaar/lorawan@v0.0.0-20200903101545-e07a1f3df89c/gps

E.g.

```go
gpsTime  := gps.Time(time.Now()).TimeSinceGPSEpoch()

```

vfylyk, 2020-09-29:
Thanks @brocaar, I'll give it a try when a get a moment.

PR #181 (2020-10-09): Basic Station time synchronization.
Closes #179.

Supports request and response to the ["timesync" message type](https://doc.sm.tc/station/tcproto.html#synchronizing-pps-to-gps-time) of the LNS protocol.

Tested with a Cisco IXM and the gateway time is now provided in the uplink messages.
commit 1062b1d7 (2020-10-09): Basic Station timesync request/response support. (#181)

Closes #179.
closed 2020-10-09 by brocaar