chirpstack/chirpstack-gateway-bridge #187: Get null time in uplink using basic-station

sagar-patel-sls, 2021-02-17:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## What happened?
we have enabled GPS in our basic-station gateway but we are not getting time in the uplink message.

## Could you share your log output?
[image]

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Gateway Bridge      | v3.10.0  |

sagar-patel-sls, 2021-06-23:
@brocaar 
any update on this issue? should I make PR for this?

brocaar, 2021-06-23:
@sagar-patel-sls a PR would be very welcome :) Could you check if the `gpstime` is correctly set (if this value is parsed at all)?

https://doc.sm.tc/station/tcproto.html#radio-metadata

https://github.com/brocaar/lorawan/tree/master/gps provides functions to convert the GPS time to a `time.Time` type (for setting the `time` field). I think both should be exposed in the RxInfo. Failing that, if the fake rx time is enabled, just set the `time` with `time.Now()`.

sagar-patel-sls, 2021-06-23:
`{"msgtype":"updf","MHdr":64,"DevAddr":xxxxxxxx,"FCtrl":0,"FCnt":10,"FOpts":"","FPort":2,"FRMPayload":"B3354E4C7F7C4356CE207E51AF7C4F","MIC":-433357060,"RefTime":0.000000,"DR":0,"Freq":865062500,"upinfo":{"rctx":0,"xtime":7599824395191180,"gpstime":0,"fts":-1,"rssi":-109,"snr":-2.5,"rxtime":1613033399.383087}}`

currently, basic station sending the above payload with a `GPS `value of 0. I think we need to take time from `rxtime` using the below `math` function so we can convert it to Unix time.

`sec, nsec := math.Modf(rmd.UpInfo.RxTime)`

brocaar, 2021-06-23:
Looks like that field is not documented (see my previous link). In that case it makes sense to map the `rxtime` to `time` (if not `0`) and `gpstime` (if not `0`) to `time_since_gps_epoch` instead of doing any gps time > `time.Time` conversions.

PR #196 (2021-06-30): Fix null time when GPS not available in GW
Fixes #187 

@brocaar Please review this implementation.
closed 2021-06-30 by brocaar