chirpstack/chirpstack-gateway-bridge #194: [3.11.0-r2 / Kerlink iFemtoCell] Failed to connect to MQTT broker (OpenVPN)

KlectikDev, 2021-06-07:
Hello,

I open this issue because the version 3.11.0-r2 of Chirpstack Gateway Bridge is unfortunately unusable on our Kerlink iFemtoCell and Kerlink iFemtoCell-evolution gateways.

Here is a description of the involved setup : 
- Several Kerlink iFemtoCell gateways are installed at different places, accessing the Internet via different networks.
- The gateways are all connected to an OpenVPN server, publishing the route giving network access to the server running the MQTT broker.

Here are the Chirpstack Gateway Bridge startup logs, at maximum logging level. Note that the following informations have been replaced : 
- {GATEWAY_ID}
- {BROKER_IP}
- {BROKER_PORT}
```
INFO[0000] starting ChirpStack Gateway Bridge            docs="https://www.chirpstack.io/gateway-bridge/" version=3.11.0
INFO[0000] backend/semtechudp: starting gateway udp listener  addr="127.0.0.1:1700"
INFO[0000] integration/mqtt: gateway id provided by authentication method  gateway_id={GATEWAY_ID}
DEBU[0000] backend/semtechudp: cleanup gateway registry
INFO[0000] integration/mqtt: setting last will and testament  gateway_id={GATEWAY_ID} topic=gateway/{GATEWAY_ID}/state/conn
DEBU[0000] [client]   Connect()                          module=mqtt
DEBU[0000] [store]    memorystore initialized            module=mqtt
DEBU[0000] [client]   about to write new connect msg     module=mqtt
ERRO[0000] [client]   dial tcp {BROKER_IP}:{BROKER_PORT}: connect: connection refused  module=mqtt
WARN[0000] [client]   failed to connect to broker, trying next  module=mqtt
ERRO[0000] [client]   Failed to connect to a broker      module=mqtt
DEBU[0000] [store]    memorystore closed                 module=mqtt
ERRO[0000] integration/mqtt: connection error            error="network Error : dial tcp {BROKER_IP}:{BROKER_PORT}: connect: connection refused"
DEBU[0002] [client]   Connect()                          module=mqtt
DEBU[0002] [store]    memorystore initialized            module=mqtt
DEBU[0002] [client]   about to write new connect msg     module=mqtt
ERRO[0002] [client]   dial tcp {BROKER_IP}:{BROKER_PORT}: connect: connection refused  module=mqtt
WARN[0002] [client]   failed to connect to broker, trying next  module=mqtt
ERRO[0002] [client]   Failed to connect to a broker      module=mqtt
DEBU[0002] [store]    memorystore closed                 module=mqtt
ERRO[0002] integration/mqtt: connection error            error="network Error : dial tcp {BROKER_IP}:{BROKER_PORT}: connect: connection refused"
...
```
The last 8 lines repeat endlessly.

Here is the configuration used : 
```
...

[integration]
marshaler="protobuf"

  [integration.mqtt]
  event_topic_template="gateway/{{ .GatewayID }}/event/{{ .EventType }}"
  command_topic_template="gateway/{{ .GatewayID }}/command/#"
  max_reconnect_interval="1m0s"
  terminate_on_connect_error=false

  [integration.mqtt.auth]
  type="generic"

    [integration.mqtt.auth.generic]
    servers=["tcp://{BROKER_IP}:{BROKER_PORT}"]
    qos=1
    clean_session=false
    client_id="{GATEWAY_ID}"
...
```
I tried to run the software with and without the use of the new connection state messages feature.
I also tried to run the software omitting the `qos`, `clean_session`, and `client_id` parameters.

This configuration works perfectly on CSGB v3.10.0-r2.

We use the Kerlink firmware v4.3.3, the latest one up to this date.

brocaar, 2021-06-29:
> network Error : dial tcp {BROKER_IP}:{BROKER_PORT}: connect: connection refused

That is a network error. The ChirpStack Gateway Bridge is unable to connect to `{BROKER_IP}:{BROKER_PORT}` because the connection is refused (firewall?).

KlectikDev, 2021-06-29:
> > network Error : dial tcp {BROKER_IP}:{BROKER_PORT}: connect: connection refused
> 
> That is a network error. The ChirpStack Gateway Bridge is unable to connect to `{BROKER_IP}:{BROKER_PORT}` because the connection is refused (firewall?).

That was not an actual network problem, as I tried the previous version and this one back and forth in the exact same setup.

However, I solved this issue by checking the new version of your documentation out and by installing the generic KerOS version you put out at https://artifacts.chirpstack.io/vendor/kerlink/keros-gws/.

Thank you for your reply.

closed 2021-06-29 by KlectikDev

brocaar, 2021-06-29:
Ah, then this issue was probably related to some changes that were made in the latest KerOS firmware. Some paths were updated and the latest generic KerOS package does reflect these changes. Glad to hear the latest package solves the issue :)