chirpstack/chirpstack-gateway-bridge #195: Binary doing nothing from version 3.9 onwards

r41d, 2021-06-18:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## What happened?
Tried running latest version (3.11) on Dragino LG308 and the binary does not produce any output and consumes a lot of CPU

## What did you expect?
Some output of any kind

## Steps to reproduce this issue

Steps:

1. SSH into Dragino LG308
2. Install chirpstack-gateway-bridge ipk file via opkg
3. Execute `/opt/chirpstack-gateway-bridge/chirpstack-gateway-bridge --help`
4. Binary runs endlessly, consuming ~1/8 of RAM and grilling the CPU (killed it after an hour)

## Could you share your log output?

Not really, because the binary doesn't output anything.

Version 3.8 is working! The erroneous part must have been introduced in version 3.9.

## Your Environment

I don't think this is relevant here, because I cannot even get the binary to start up

brocaar, 2021-06-21:
Hi @r41d please note that I needed to perform some workarounds to make the binary fit and work at all on the LG308 gateway, due to the limited amount of flash memory. It could be that due to changes / compiler chain updates the size increased which broke support for the LG308 gateway.

I don't think there is an easy fix for this, other than implementing a slimmed down version of the ChirpStack Gateway Bridge. I am considering something like this in the future, but I have first a couple of other items I want / need to work on.

r41d, 2021-06-21:
Hmm, I don't really observe an change in file size from 3.8 to 3.9 :thinking: 
From looking over the git history betweeen 3.8 and 3.9, I think the only commit that had significant changes was the one called "Support multi-downlink commands and acks." and from briefly looking into it I saw that the `patrickmn/go-cache` library was added, my best guess would be that it could have something to do with that

brocaar, 2023-01-26:
Please see https://www.chirpstack.io/docs/chirpstack-mqtt-forwarder/install/dragino.html. This is a re-implementation of the ChirpStack Gateway Bridge, but then a lot more compact and only intended to install on the gateway.

closed 2023-01-26 by brocaar