chirpstack/chirpstack-gateway-bridge #203: Add ZeroMQ Pub/Sub as Integration Option

cBiscuitSurprise, 2021-11-05:
> I had created this with the wrong account. Using this one going forward.

- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## Summary

## What is the use-case?

I'm in the process of deploying a full LoRa stack at the edge. We already have an MQTT**S** broker at the edge, but it would be a hassle to configure this bridge to access that broker (certificate generation and dissemination at scale would be more of a hassle than it's worth). I'd also like to avoid having two brokers on the same server as all I need is a means to forward packets to the local Network Server.

## Implementation description

Example concept configuration:
```toml
# Integration configuration.
[integration]
type="zmq"
# Payload marshaler.
#
# This defines how the MQTT payloads are encoded. Valid options are:
# * protobuf:  Protobuf encoding
# * json:      JSON encoding (easier for debugging, but less compact than 'protobuf')
marshaler="protobuf"

  # ZMQ integration configuration.
  [integration.zmq]
  event_url="ipc:///tmp/gatewaybridge_event"
  command_url="ipc:///tmp/gatewaybridge_command"
```

## Can you implement this by yourself and make a pull request?

I intend to implement and create a PR myself assuming it's welcomed. I would also be creating a complementary PR in the `chirpstack-network-server` repository implementing the other side of this integration.

brocaar, 2021-11-06:
For now, I would like to park this. While I think this would be an interesting feature, I'm currently in the process of refactoring quite some code (nothing public yet and I can't give an ETA for this, but I believe it will also help for edge deployments).

For edge deployments, I think it would be even better to bypass the ChirpStack Gateway Bridge entirely and add a Concentratord backend (which already uses ZMQ) to the ChirpStack Network Server. Again, I would like to add this after the refactoring work but I'm very interested in adding this as a feature.

cBiscuitSurprise, 2021-11-06:
Sounds good. I do have concentratord running at the edge already, but wasn't familiar enough to know if bypassing this bridge was practical. Thanks for building such an awesome set of projects! If I can help at all, let me know.

closed 2021-11-06 by cBiscuitSurprise