chirpstack/chirpstack-gateway-bridge #212: [Build] missing go.sum entries

mullerch, 2022-09-06:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## What happened?

A clean build fails:

```
go: github.com/spf13/viper@v1.12.0 requires
        github.com/sagikazarmark/crypt@v0.6.0: missing go.sum entry; to add it:
        go mod download github.com/sagikazarmark/crypt
```

This is just the fist of multiple missing entries.

## What did you expect?

Obviously, I expect it to build ;-)

## Steps to reproduce this issue

Steps:

1. Build from clean

Looking the log I guess it has been introduced by d6b046ec8fd220923442d9deb0f1e8eefe92606d.

## Workaround

Run `go mod tidy` before the build.

You may not get the same versions of the dependencies as the tested ones.

brocaar, 2022-09-06:
> Build from clean

Which works for me :-) Are you using the provided docker-compose environment to build the ChirpStack Gateway Bridge? Note that I reverted back the `go 1.18` to `go 1.17` as I was running into a similar issue when trying to compile with an older Go compiler.

mullerch, 2022-09-06:
Oh you are right, it works with 1.18 bu not 1.16. I did not expect a change in the go version to have such a significant change in the build system. Sorry for that.

closed 2022-09-06 by mullerch