chirpstack/chirpstack-gateway-bridge #213: Gateway Bridge not starting after reboot under Multitech v6.0.0

rekcaiot, 2022-09-29:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## What happened?

After updating the mPower Firmware version from 5.3.3 to 6.0.0 the soft reboot via cli of the gateway no longer leads to the start of the Gateway Bridge after boot. When doing a hard shutdown and restart the Gateway Bridge starts like expected.

## What did you expect?

The Gateway Bridge starts after soft "reboot" under mPower 6.0.0.

## Possible solution

When having a look into the original chirpstack-gateway-bridge.init file of the Gateway Bridge the LSB Header is present:
\### BEGIN INIT INFO
\# Provides:          chirpstack-gateway-bridge
\# Required-Start:    $all
\# Required-Stop:     $remote_fs $syslog
\# Default-Start:     2 3 4 5
\# Default-Stop:      0 1 6
\# Short-Description: ChirpStack Gateway Bridge
\### END INIT INFO

The vendor specific one for Multitech Conduit doesn't have the LSB Header.
When adding this header manually and reinitilizing the service, the reboot works like expected.

The Bridge also starts after reboot when adding the LSB Header under mPower 5.3.3.

| Component           | Version |
| --------------------| ------- |
| Gateway Bridge      |    chirpstack-gateway-bridge_3.14.0-r1_arm926ejste     |
| Multitech Gateway | MTCAP2-868-042A |
| Multitech Firmware Version | mPower 6.0.0 |

brocaar, 2022-09-29:
To take away any assumptions and making it easier to reproduce, could you describe the exact commands / steps for:

* "soft reboot via cli" (do yo mean `reboot`)?
* hard shutdown and restart (do you mean using `shutdown` and then power-cycling the gateway)?

brocaar, 2022-10-04:
I've been able to reproduce this, but in my case adding the above header did not solve it. What I think is that the ChirpStack Gateway Bridge starts too early (e.g. before network setup).

What I did see is that the mPower firmware provides Monit. I will update the `.ipk` such that:

* Monit will be enabled on CGB `.ipk` install
* CGB process will be monitored by Monit (in case of failure, it will be automatically restarted)

commit 6cb0d357 (2022-10-04): Update Multitech Conduit packaging to make use of Monit.

Fixes #213.
closed 2022-10-04 by brocaar