chirpstack/chirpstack-gateway-bridge #214: Gateway bridge on  Kerlink iStation only forwards a few messages after bootin, then stops publishing to MQTT

ldorigo, 2022-10-05:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## What happened?

I already posted about this on the community forum (https://forum.chirpstack.io/t/kerlink-istation-heartbeat-messages-are-sent-to-the-gateway-but-most-uplink-messages-are-lost/14922/6), but I'm now relatively confident it's a bug in the Gateway Bridge so I'm opening an issue here. Already explained the issue in detail there so refer there for more detail.

The forwarder appears to timeout on most messages it forwards to the gateway-bridge. Some messages go through, however - notably, just after the gateway is rebooted. Relevant logs where one packet goes through and one does not:

```
2022-10-05T11:26:53.969143+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:26:53.968844464+02:00" level=debug msg="backend/semtechudp: received udp packet from gateway" addr="127.0.0.1:37642" protocol_version=2 type=PushData                   
2022-10-05T11:26:53.975921+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:26:53.975689768+02:00" level=debug msg="backend/semtechudp: sending udp packet to gateway" addr="127.0.0.1:37642" protocol_version=2 type=PushACK                       
2022-10-05T11:26:55.817461+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:26:55.817273733+02:00" level=debug msg="[pinger]   ping check 4.246189219" module=mqtt                                                                                  
2022-10-05T11:26:56.989257+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:26:56.988963004+02:00" level=debug msg="backend/semtechudp: received udp packet from gateway" addr="127.0.0.1:37642" protocol_version=2 type=PushData                   
2022-10-05T11:26:56.991936+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:26:56.991619789+02:00" level=info msg="integration/mqtt: publishing event" event=up qos=0 topic=eu868/gateway/7076ff0056070a51/event/up uplink_id=14509                 
2022-10-05T11:26:56.992987+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:26:56.992670502+02:00" level=debug msg="[client]   enter Publish" module=mqtt                                                                                           
2022-10-05T11:26:56.994727+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:26:56.993680547+02:00" level=debug msg="[client]   sending publish message, topic: eu868/gateway/7076ff0056070a51/event/up" module=mqtt                                 
2022-10-05T11:26:56.997848+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:26:56.997521717+02:00" level=debug msg="[net]      obound msg to write 0" module=mqtt                                                                                   
2022-10-05T11:26:56.998758+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:26:56.995323286+02:00" level=debug msg="backend/semtechudp: sending udp packet to gateway" addr="127.0.0.1:37642" protocol_version=2 type=PushACK                       
2022-10-05T11:26:57.004184+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:26:57.004024005+02:00" level=debug msg="[net]      obound wrote msg, id: 0" module=mqtt                                                                                 
2022-10-05T11:26:57.004455+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:26:57.004333019+02:00" level=debug msg="[net]      outgoing waiting for an outbound message" module=mqtt                                                                
2022-10-05T11:27:00.004971+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:00.00453268+02:00" level=debug msg="backend/semtechudp: received udp packet from gateway" addr="127.0.0.1:37642" protocol_version=2 type=PushData                    
2022-10-05T11:27:00.005747+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:00.005544391+02:00" level=debug msg="backend/semtechudp: sending udp packet to gateway" addr="127.0.0.1:37642" protocol_version=2 type=PushACK                       
2022-10-05T11:27:00.817333+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:00.817109034+02:00" level=debug msg="[pinger]   ping check 3.812699679" module=mqtt                                                                                  
2022-10-05T11:27:00.817607+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:00.817484051+02:00" level=debug msg="[pinger]   keepalive sending ping" module=mqtt                                                                                  
2022-10-05T11:27:00.849633+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:00.849382798+02:00" level=debug msg="[net]      startIncoming Received Message" module=mqtt                                                                          
2022-10-05T11:27:00.849965+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:00.849830151+02:00" level=debug msg="[net]      startIncomingComms: got msg on ibound" module=mqtt                                                                   
2022-10-05T11:27:00.850251+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:00.850125831+02:00" level=debug msg="[net]      startIncomingComms: received pingresp" module=mqtt                                                                   
2022-10-05T11:27:00.853485+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:00.853103963+02:00" level=debug msg="[net]      logic waiting for msg on ibound" module=mqtt                                                                         
2022-10-05T11:27:01.557169+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:01.55685516+02:00" level=debug msg="backend/semtechudp: received udp packet from gateway" addr="127.0.0.1:35872" protocol_version=2 type=PullData                    
2022-10-05T11:27:01.557680+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:01.557488188+02:00" level=debug msg="backend/semtechudp: sending udp packet to gateway" addr="127.0.0.1:35872" protocol_version=2 type=PullACK                       
2022-10-05T11:27:01.559447+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:01.559149262+02:00" level=debug msg="integration/mqtt: set gateway subscription" gateway_id=7076ff0056070a51 subscribe=true                                          
2022-10-05T11:27:01.559752+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:01.559608616+02:00" level=debug msg="integration/mqtt: gateway subscription set" gateway_id=7076ff0056070a51 subscribe=true                                          
2022-10-05T11:27:03.017350+02:00 klk-wiis-070A51 chirpstack-gateway-bridge[1184]: time="2022-10-05T11:27:03.01703189+02:00" level=debug msg="backend/semtechudp: received udp packet from gateway" addr="127.0.0.1:37642" protocol_version=2 type=PushData              
``` 

Packet  Forwarder:

```
:26:53.964734+02:00 klk-wiis-070A51 lorafwd[1216]: <7>  gwmp.send:  02 AD 37 00 70 76 FF 00 56 07 0A 51 7B 22 72 78 70 6B 22 3A 5B 7B 22 61 65 73 6B 22 3A 30 2C 22  |..7.pv..V..Q{"rxpk":[{"aesk":0,"|                                                                                                                      
:26:53.965841+02:00 klk-wiis-070A51 lorafwd[1216]: <7>  gwmp.send:  62 72 64 22 3A 33 2C 22 63 6F 64 72 22 3A 22 34 2F 35 22 2C 22 64 61 74 61 22 3A 22 22 2C 22 64  |brd":3,"codr":"4/5","data":"","d|                                                                                                                      
:26:53.966435+02:00 klk-wiis-070A51 lorafwd[1216]: <7>  gwmp.send:  61 74 72 22 3A 22 53 46 37 42 57 31 32 35 22 2C 22 64 65 6C 61 79 65 64 22 3A 74 72 75 65 2C 22  |atr":"SF7BW125","delayed":true,"|                                                                                                                      
:26:53.966902+02:00 klk-wiis-070A51 lorafwd[1216]: <7>  gwmp.send:  66 72 65 71 22 3A 38 36 37 2E 37 2C 22 6A 76 65 72 22 3A 32 2C 22 6D 6F 64 75 22 3A 22 4C 4F 52  |freq":867.7,"jver":2,"modu":"LOR|        [... cut ...]
Configuration files:

[gist:556ad5718765129eeb7a811b362cb5b8 · GitHub](https://gist.github.com/556ad5718765129eeb7a811b362cb5b8)

[gist:0ee5e5edaaeb26fdf15b6372928605be · GitHub](https://gist.github.com/0ee5e5edaaeb26fdf15b6372928605be)

| Component           | Version |
| --------------------| ------- |
| Application Server  | v4 (docker) |
| Gateway Bridge      |   v4.1 (published 4th of october, bug was also present in previous version)     |
| Forwarder | Semtech UDP built-in to kerlink gateway |

brocaar, 2022-10-06:
Which firmware version are you using on your gateway? If it is not the latest version, could you try upgrading the firmware and try to reproduce the issue?

ldorigo, 2022-10-20:
Hi, sorry for the slow response and thanks for the suggestion - we were finally able to upgrade the firmware version, and it seems to work now.

closed 2022-10-20 by ldorigo